travisgoodspeed/md380tools #137: Option to edit programmed channel...

dg9vh, 2016-05-27:
... to modify "Contact Name" in codeplug section "Channels Information". To be placed in "Utilities"  "program radio" as point after "Repeater Slot".

Close if functional

travisgoodspeed, 2016-05-27:
This description is a bit too terse.  Could you explain the issue more clearly?

dg9vh, 2016-05-27:
Trying to describe it step by step:

- Adding menupoint "Utilities"-"program radio"-"Change Contact Name"
- load all contacts into a list marked as "group call"
- select current Contact Name selected in codeplug as default-contact for programmed channel
- after selecting other contact write this into channel information

travisgoodspeed, 2016-05-27:
The feature is certainly reasonable to add.  Any volunteers to write the patch?

mach327, 2016-09-27:
If I understand correctly, the idea is to be able to edit a digital channels contact/txgroup from the front panel. That would seem to be a critical feature for programming digital channels from the front panel.

travisgoodspeed, 2016-10-01:
Merged into #370. Reopen only if there is active progress in adding such a feature.

closed 2016-10-01 by travisgoodspeed