travisgoodspeed/md380tools #143: Decoding issue

ct2iww, 2016-06-07:
Hello everyone, 

I have an issue with my Retevis RT3 that seems to be quite serious, given that many folk are starting to report the same problem, in some foruns, namely TYT and MMDVM, on yahoo groups.

With the latest firmware loaded via MD380 tools, my radio seems to be working fine but in the middle of a QSO,it will stop decoding the other party´s transmission. The green led remains lit and I can hear the transmission on DSD+ and RTL dongle, but the RT3 remanais mute. 

To regain the conversation back, I have to manually switch the channel selecting knob back and forth. At that point, the other party´s transmission can be decoded. This happens regardless I am talking to someone locally, via Brandmeister or radio brand ( so far, it happens with both Moto, Hytera and TYT radios).

This issue happens mostly on QSO´s (not while listening only) and it seems to be triggered by the display going blank, after the 15 seconds expire.

Loading the manufacturers firmware (2.034) the problem goes away, even if the display goes to sleep. So far, by reading other MD 380 owners problems, I seem to be the only one tying the issue to the alternative firmware. 

However, it might be a bit early to be sure of that, since this just started being discussed. Suffice to say, thought, it seems that some of these radios are poorly aligned from factory (happened to a friend of mine with an RT8, had to be returned).

So, there it is. And now?

Paulo, CT2IWW

aeickho, 2016-06-07:
Paulo, thanks for your report.

me is the behavior already noticed but I use the radio so few that I could not reproduce it.

when I ask other user I heard that this behavior even with the original firmware occurs.

let us wait for more "unhappy" reports.

or please disable the md380 additional functions.

aeickho, 2016-06-07:
#140 is one possible location for this problem, maybe

travisgoodspeed, 2016-06-07:
Paulo, please check two things for us:

1. Does the problem occur with stock 2.032 firmware?
2. Does the problem occur when running our firmware with UsersCSV disabled?

Cheers from Seattle,
KK4VCZ

ct2iww, 2016-06-07:
Hi Travis,

I have loaded stock 2.034 at the moment and that problem does no occur. The
problem occurs with the CSV list turned on or off, under the patched
firmware. I also uploaded a well known codeplug for our country that is in
widespread use and with no reports of problems but the decode failure
persists.

I only have access to a local MMDVM repeater but since the other local
radios seem to work with it with no problems, I am less inclined to blame
it for this issue.

During the week I have very few local ops to talk to and do some further
testing wtih. However, since we have a three day weekend coming up, I´ll
reload the patched firmware and try to do some more tests. Lets see also
what comes form the other forums, I think it will give me a better idea if
is a patched firmware issue only or a more widespread one.

Cheers from Amarante,

Paulo, CT2IWW

dg9vh, 2016-06-13:
Hi,

after my vacation I'll start comparing the receiving behavior of original 2.032 and patched firmware on different environments that are a homebrew MMDVM-Repeater, a DMRPlus-Repeater, a DVMEGA and a DV4Mini... if both behave the same, I would compare patched with 2.034 and patched with 3.008 (firmware from this week).

After this, I would post my results here. But stay tuned, I'm on vacation from Wednesday for 11 days...

g7npw, 2016-06-16:
I get this same problem also & can confirm 2.034 does fix it with MMDVM Repeaters.

g7npw, 2016-06-19:
Have been testing this again this weekend and have also seen same problem on a Hytera Repeater on the Brandmeister network with firmwares stock <2.032

Stock >2.034 seems to fix issues including 3.008.

Disabling battery saver does not help either.

Dom G7NPW

rct, 2016-06-19:
@g7npw - your notation is a little confusing, instead of `<` (less then) do you mean `<=` as in the problem occurs in 2.032 and earlier, but is resolved in 2.034 and later?

g7npw, 2016-06-19:
Sorry, yes I have found the problem to occur in 2.032 and earlier unmodified firmwares. And also the modified versions have it too. 2.034 and the 3.008 do not seem to have the problem.

I have seen it on MMDVM & Hytera repeaters connected to Brandmeister network. Not tried any others.

It is more likely if the repeater has a long hang time as when carrier drops it helps the radio recover and decoded again.

aeickho, 2016-07-09:
@dg9vh you have news about this behavior and your analyses?

g7npw, 2016-07-09:
I'm still waiting for others to experience the same issue and report back. A modified firmware based upon newer tyt firmware would be good.

dg9vh, 2016-07-10:
My 2nd ht was on vacacion with my step-father but back this week so I go on testing this week.

At the moment my experiences with 2.032 are, that both have this problem but not at the same times. It's more like random.

awasser1, 2016-07-18:
like 10 of us in Texas Brandmeister group tg 3148 have the problem every day.  we get green light have to push PTT.  Its random enough to be annoying.  It happens everyday to us.

travisgoodspeed, 2016-07-18:
Does anyone have packet captures that could explain exactly what goes wrong when this bug is triggered?

With issues #175 and #181, we are in the process of moving away from 2.032 to more modern firmware revisions.  This will take some time, but it will have the happy benefit of fixing this issue.

awasser1, 2016-07-18:
I'll check with someone in group that knows how to do this. Will advise. Thanks
  
  
 Sent from   the future (https://itunes.apple.com/app/apple-store/id922793622?pt=814382&mt=8&ct=future)

g7npw, 2016-07-18:
Is your repeater on Brandmeister in Texas Homebrew MMDVM or a commercial Motorola/Hytera unit?

2.34 , 3.008 & 3.020 fixed this issue with me. (Think I got the numbers right) so would be great to base the modified firmware on these versions one day.

awasser1, 2016-07-18:
Actually I am in CT.  the rest are in TX.   Most use the DV-Mega.   I use both DV-Mega and Brandmeister Repeater in CT.  and a couple have DV4Mini and I have one too.  Exhibited the behavior on all.  

 

 

Alec

 

We like

awasser1, 2016-07-18:
Do I turn on debug on my md380 tools to get the log for the next time?  I’m not sure how to get a log from this am?

iw9grl, 2016-07-19:
Hi I confirm that also with 3.020 the dropped mid-transmission with Retevis rt3 is present. 73

iw9grl, 2016-07-24:
Hi dear, any news about ?

tnx
73

dg9vh, 2016-07-24:
From my side not really...
> 

-- 
Amateurfunk ist das, was uns verbindet! Bleiben wir in Kontakt!
Ortsverband Q05, Völklingen

aeickho, 2016-08-27:
Please report if this problem occurs also with d13.020

kirkmawa, 2016-09-16:
Not sure if this is related, but on an MD380/no GPS/old vocoder with md380tools firmware built last night, if the radio loses the stream at any point mid-call, it freezes like described earlier in the thread and additionally keeps playing the same audio sample from the buffer until I switch channels. I had promiscuous mode enabled, and this happened with UsersCSV enabled and disabled. 

73 de KN9FOO

awasser1, 2016-09-16:
That’s most likely not phone problem.   Assuming you are on Brandmeister and someone has two hotspots active and they put on on TS 1 and TS 2 it causes that playing over and over.  Long and short of it…not your Radio.   The missing audio issue was fixed when the firmware started using a new release

 

 

Alec

N1AJW

travisgoodspeed, 2016-09-16:
Close this issue if we someone can confirm that, as N1AJW claims, it's not the radio.  Also close it if we can confirm that 13.020 is unaffected.

-- YU/KK4VCZ

closed 2016-10-18 by travisgoodspeed

awasser1, 2016-10-18:
Yes it's fixed doesn't happen anymore !  Thanks for all your hard work 

Sent from Alec's iPhone 7