travisgoodspeed/md380tools #146: Apply patches to 3.008 Firmware

M0IVQ, 2016-06-10:
Firmware version 3.008 is now available which according to reports gives scrolling text on display. Is it worth looking into applying the patches to this version?

Close when patches can be applied to 3.008 firmware or if not feasible to do so.

bilbolodz, 2016-06-10:
Where is it available to download?

M0IVQ, 2016-06-10:
It's been uploaded to the TYT TYTERA yahoo group and apparently available on tyt888.com but the latter doesn't seem to be working at the moment so can't verify it at the moment. Have installed on my MD-380 buy haven't used it yet other than checking in radio info.

travisgoodspeed, 2016-06-11:
The major new feature of 3.008 is support for GPS on some (not all) of the MD390 devices.  It might be easier for us to backport GPS functionality than to relocate all of our patches to the new firmware.

M0IVQ, 2016-06-11:
From what I've seen, the GPS version has firmware version of D013.009 which is completely incompatible with MD380 and early non GPS MD390

travisgoodspeed, 2016-06-11:
Any luck finding a copy of the updater?

The non-GPS MD390 runs our firmware fine.  I'll pick up a GPS model as soon as they appear on Amazon.

M0IVQ, 2016-06-11:
https://onedrive.live.com/redir?resid=8F7CA0DBE9FAF990!140550&authkey=!AJ0uAGwMtLkuzEI&ithint=file%2cbin

dg9vh, 2016-06-14:
Infos about changelog (source: tytera yahoo group):
```
Some notes on what you can expect in this latest version of firmware, an YES you’ll want to upgrade to this if you can…

 

1.       CPS, v1.30 still works with programming this radio!

2.       You can upgrade using the v2.034 image zip tool, just replace the binary with the later v003.008 binary.

3.       Here are some notable features you’ll see with this firmware update:

a.       Scrolling Text Display Names

b.       Complete front panel programmability, including the following options for DMR channels:

RX Frequency

TX Frequency

Channel Name

Time Out Timer

CTCSS/DCS

Color Code

Repeater Slot

Tx Contact

RX Group List

 
```

bilbolodz, 2016-06-15:
Does these new version usable also with VHF band radio?

travisgoodspeed, 2016-06-15:
Prior versions ran on both the UHF and VHF model.  I don't expect it to give you any trouble.

travisgoodspeed, 2016-06-16:
The additional front-panel programming will be handy, but I'm still hesitant to rewrite all of the symbols to a new version.  Close this issue when we've backported (1) GPS and (2) the additional front panel programming features.

Anything else from this update worth paying attention to?

n7eal, 2016-06-17:
I just got a md-380 and it has D013.009 
definitely a 380, not a 390.
it's from a ebay dealer in Los Angles that sells a lot I think, so it prob fresh off the boat.
I've had a cs700 for 1.5 years, and got a 380 to play with this firmware tools
  I'm hoping to find the bin for D013.009 in case I need to go back to stock.
thanks for the great tools!

M0IVQ, 2016-06-17:
There's a note with the 3.008 firmware saying not to try it with a radio with version Dx1x.xxx firmware so it's possible theres a hardware difference with this radio making original firmware's incompatible.

bilbolodz, 2016-06-17:
I've successfully loaded 3.008 into VHF band radio. I've NOT checked transmit/receive (no working rep in my area right now) but at least it's not a brick.

n7eal, 2016-06-20:
I did find a D013.009 bin on the md380 yahoo group if anyone is interested in checking it out.
The person said it worked on there 380 that the older fw did not.
-n7eal

M0IVQ, 2016-06-20:
It would be interesting to see what the difference between D013.009 & D003.008 is - as they are functionally the same, it may help pinpoint the difference required for 'new' and 'old' MD380/390's.

I've tried using the radiff2 -C command on unwrapped bin files but am not getting any output from it. Can anyone point me in the right direction as I'm new to r2?

ea8cwb, 2016-06-20:
Hi. I have a tytera md380 running with TYT-MD390-D013.009.bin firmware. I had a problem because i try to update the firmware and no success. My walkie got crazy in restarting loopings. So i contacted with a lot of people, even with Travis. Noboby knew whats happening because every diferent version of firmware i tried to update got the same results... reset... reset... reset.. i didn´t got enter to the main menu. Bue a guy, sent me the firmware D013.009.bin wich is published in http://ea8cwb.weebly.com/dmr.html i decided to try to update the firmware one more time and... success. Walkie is running lovly, but no accepts any other firmware... 

I hope my experience will help to fix the problem and patch the new one that affects to a lot of new guys who will buy the new tyteras.

See you soon,
EA8CWB, Eduardo.

travisgoodspeed, 2016-06-20:
1. We need to get a few of these troublesome radios for testing and porting. Does anyone know a vendor for them?
2. We can fix this either by automated symbol migration (harder than it sounds) or by patching the older firmware to work around whatever hardware difference exists.
3. It can't hurt to ask Tytera for the new schematics, in case they documented the change.

-Travis

ea8cwb, 2016-06-20:
I know one! Where i bought mine:
http://s.aliexpress.com/aAJfuU3I

Un saludo,

Eduardo Díaz.

n7eal, 2016-06-20:
I got mine on Ebay from a dealer in Los Angles California called 'letsgetready' that has the D013.009 radios about $100 USD
not sure if the shipping from the US works out for you.
n

travisgoodspeed, 2016-06-22:
Splitting the discussion of D013.009 and the reboot bug to #155.

travisgoodspeed, 2016-07-08:
We won't be targeting 3.008, but I have hardware coming and we'll either target 13.009 for new units or patch 2.032 to be compatible with the new hardware.  See #155 for that discussion.

Closing issue.

closed 2016-07-08 by travisgoodspeed