travisgoodspeed/md380tools #147: Scrolling zone with the up and down arrow during receiving

IZ6MAF, 2016-06-11:
Zone selection during reception.
Using the transceiver, traveling with the car, we have continually the need to move from one zone ( 1st repeater ) to another zone ( 2 ° repeater ) to a third zone and so on (only in Italy we have more than 69 different zone). 
The current procedure for changing the zone  is long rather laborious, and must be done by pressing the following keys: "confirm" -> Arrow down -> Arrow down -> "confirm" -> "Multiple Arrow Up & Down to zone selection" -> "confirm".
As when the unit is receiving the arrow buttons on the keyboard have no function, please can you add a tool that permits to us the zone selection function pushing the keyboard arrow buttons down and up during the reception?
Thank you very much
'73 Paolo IZ6MAF

M0IVQ, 2016-06-11:
One of the side buttons can be programmed to increment the zone on stock firmware but only in one direction.

IZ6MAF, 2016-06-11:
Yes I know, but to go up one zone I need to go down 69 zone, it is not really practical !!!

IZ6MAF, 2016-06-11:
Yes I know, but to go up one zone i must go down 69 zone, it's not really pratical

M0IVQ, 2016-06-11:
Sorry, my comment was meant for informational purposes as may assist with patching to enable scrolling up and down rather than just down.

travisgoodspeed, 2016-10-01:
Merged into #370. Reopen only if there is active progress in adding such a feature.

closed 2016-10-01 by travisgoodspeed