travisgoodspeed/md380tools #148: Completely disable encryption on firmware

n0p, 2016-06-14:
In some countries (like Spain) it's strictly forbidden to use radios with encryption capabilities (even if you don't enable it)

So it would be great to add a makefile switch to disable or bypass the encryption features of the radio to use it on these countries.

travisgoodspeed, 2016-06-14:
Is this a law that's enforced?  Does it apply to the equipment of amateurs, or only unlicensed operators?

From what I recall, the crypto function is easy enough to disable, but it needs to encrypt one block correctly as part of the power on self test. Bypassing that check and hooking the crypto function will do what you want. 

I'll post more notes on how the crypto works this afternoon.

Cheers from Samland,
-Travis

n0p, 2016-06-14:
It's an old outdated law that's not enforced, but there's a lot of drama about DMR being illegal because of this, so I thought this would be an easy to achieve and drama-stopping feat, but not a prioritary one.

Cheers from Spain :P

gamo73uk, 2016-06-22:
I'm glad this was brought up. I'm UK based but was planning on taking my RT3 to Spain when I go in October. If there's a chance some grumpy border control person will take it away when I get there, perhaps I won't!

geiraage, 2016-06-22:
Hey folks.

Nice work you are all doing.  I am a newbie, so I need to ask.

I got 2 UHF TYT MD-380 and 2 VHF TYY MD-380.

I recently updated my 2 UHF radios with the firmware D003.09. But I
discovered that both of the 136-174 Mhz VHF radios had firmware  D013.009.

Is this D01 a description for VHF? I assume that both radio types uses the
same FW, and the experimental FW can be used on both?

Have a nice summer
Regards

Geir

n0p, 2016-06-22:
@gamo73uk no, they are not going to confiscate anything on the border, the border agents do not check this kind of things and the only problem we are having with this is on some regional telco administrations refusing to give DMR repeater installation permits and also some inter-amateur-associations drama

@geiraage I don't think this is the appropriate place for this kind of question. Maybe you can open a new issue asking for compatibility with the VHF models

travisgoodspeed, 2016-07-08:
@geiraage Compatibility with 13.009 radios is being tracked in #155.

Closing this issue as wontfix, but if you'd like to try your own hand at it, see https://github.com/travisgoodspeed/md380tools/blob/master/applet/src/aes.c

closed 2016-07-08 by travisgoodspeed

n0p, 2016-07-11:
Looking at
https://github.com/travisgoodspeed/md380tools/blob/master/applet/src/aes.c

The patch could be as simple as

> char *aes_cipher_hook(char *pkt){
>
>
>> #ifdef Disable_Encryption
>
>  return pkt;
>
> #endif
>
>
>
>  }
>
>
Am I right?

travisgoodspeed, 2016-07-11:
Three days ago, that wouldn't have worked, because the copy protection required AES to compute correctly. Now that I've disabled the copy protection, you can certainly remove that line to break encryption.

n0p, 2016-08-08:
Well, patching the hook function to avoid calls to aes_cipher() does break encryption, but it does not disable it.