travisgoodspeed/md380tools #154: Basic Privacy Crack Option - Is it a possibility?

gamo73uk, 2016-06-22:
Not sure how, or even if this could be implemented, but since BP only has 255 different possible keys,  would it be possible to have an option that you can have on a side button (like monitor etc.) which switches the front panel to display BP Key 00 then  allow you to use the up/down arrow to scroll through each possible BP key on that channel/tg etc. to slowly "brute-force" BP settings? 

You can only program 16 a time in to the radio for use on channels so setting up a 255 channel list to try each combination would take ages.

Unless someone knows of a way to do it in DSD+ or the like? ;-)

travisgoodspeed, 2016-06-22:
Basic Privacy and Enhanced Privacy can both be cracked, but I've never seen them used in the field and the Tytera algorithms are very different from the Motorola ones.

See #50 for Motorola Privacy compatibility.

travisgoodspeed, 2016-07-08:
Closing this as a duplicate of #50.

closed 2016-07-08 by travisgoodspeed

abdullah612, 2017-09-17:
any guide on how to crack them