travisgoodspeed/md380tools #155: Compatibility with D013.009 Hardware

travisgoodspeed, 2016-06-22:
As previously discussed in #146, hardware that ships with D013.009 is incapable of running earlier firmware.  We need to know why, and we need to get our patches running on the new hardware.

Use this issue to discuss the new firmware, and close this issue when either (1) we have migrated our patches to D013.009 or (2) we understand the hardware differences well enough to back-port support into our fork of 2.032.

ea8cwb, 2016-06-22:
Hi Travis. Today the vendor send me a firmware to check if works. Is named like " TYT-Vocoder-MD380-D13.09.bin" and is the lastest firmware for my MD380 type radio. For older ones he sent me the named "TYT-TFT-MD380-D3.08.bin".
So i undrestand there are any hardware difference because tytera engeeners sent 2 firmwares to solve a problem. If tytera support does not know wich one works it means the are telling us clearly there are 2 diferent models of MD380 radios.
I hope this help you. 

Rename the jpg by zip extension. System does not let me upload a zip file.
Thank you.

[image]

travisgoodspeed, 2016-06-22:
I bought a Retevis RT8, but it shipped with the 2.034 firmware.  Still looking to buy hardware that needs 13.09.

--Travis

travisgoodspeed, 2016-06-22:
@ea8cwb Thanks for the firmware!  I've downloaded the archive and compared it to the 13.009 and 03.008 images from `firmware/bin/` in our repo, and the MD5 sums match.  (I've edited your comment to cut the JPEG, so that future viewers of this page don't spend their bandwidth on it accidentally.)

The next step will be to compare these two binaries, then hopefully backport the difference.

ea8cwb, 2016-06-22:
ok Travis, if you need anything i can help let me know it.

DL9LBG, 2016-06-28:
So here is my experience with a new RT-8 delivered with 13.009 FW. I updated with Travis' FW from April - after that it was rebooting every 3 seconds. So after that I reflashed the stock 13.009 - but now only the start screen is shown shortly and then a white blank screen, no three-tone-signal after turning on. When turning off the "Powering Down" screen appears for a short time. So my RT-8 is currently unusable. Is the bootloader broken?

73s - Michael DL9LBG

travisgoodspeed, 2016-06-28:
Howdy Michael,

Does your RT8 have GPS?  I get the white screen when running 13.009 on my non-GPS RT8, which runs the older (and patched) firmware without complaint. 

-Travis

DL9LBG, 2016-06-28:
Yes, it has the GPS module (that's the reason why I bought the RT-8 instead of RT-3/MD-380).

The 3-tone-signal is there - the programming cable was still plugged in, arrrgh ...

DL9LBG, 2016-06-28:
So again I gave it a try with your older firmware (v1 from January) - endless loop with the splash screen. (new RT8 with FW 13.009 and GPS included). Flashed back to 13.009 as delivered: white screen.

DO9ULI, 2016-06-29:
Hi Travis
I have here at a RT8-GPS of my friend, the same problem.
The device starts to splash screen and has all 8 seconds a bootloop.
He wanted to flash his RT3 with your experimental software and but had the RT8 connected :-(.
In Annex I send my information that I have received so far from Retevis Support.
All tests have yet led to success.

Sorry for google-translating.

best regards

73 Uli DO9ULI
[Neues Textdokument (2).txt](https://github.com/travisgoodspeed/md380tools/files/338835/Neues.Textdokument.2.txt)

ea8cwb, 2016-06-29:
Hi do9uli. Try this firmware:
http://ea8cwb.weebly.com/uploads/5/5/7/1/5571150/tytera_walking_dead_firmware_by_ea8cwb.bin
73!

Un saludo,

Eduardo Díaz.

DO9ULI, 2016-06-29:
Hi Eduardo
Thanks for the firmware.
Unfortunately, this firmware is not working well.

It is the start screen, and then only a white screen.
If I turn off the RT8 I still see the Poweroff screen.

But I can not select menu functions.
Perhaps even comes a solution in Mment I'm just not on.
Let's see what the Retevis Service still recommends.

If there is interest I can like the Files of Retevis for RT8
provide.
Latest Firmware VHF, UHF (GPS) including an empty Codeplug

best regards

Uli DO9ULI

tkerby, 2016-07-09:
I seem to have the same problems with a unit about 8 months old that was previously running 2.32 then 2.34 official releases (it's an RT3).  Since installing the experimental firmware, the unit continually reboots. I've been unable to go back to the 2.34 firmware (that I still had a copy of from when I updated) nor any other firmware either downloaded with the md380 tools or the various walking dead firmware files out there. I either get reboots or a white screen. Firmware beyond 2.32 shows my splash screen details, older firmware does not.  I've also tried applying a virgin codeplug - a difficult task as the timing needs to be before the reboot.

This makes me think something has gone wrong or been damaged in the bootloading process and may not be specific to just the newer models

travisgoodspeed, 2016-07-09:
@tkerby Your reboot issue is probably unrelated to the hardware changes, but here are some tips that might help:

1) You can install a codeplug from the bootloader by holding PTT and the button above it during a reboot.  This way, you won't need to race the reboot in order to flash your new codeplug.

2) Flashing can go wrong if the voltage isn't stable.  Try a different battery, or at least ensure that your battery is freshly charged.

tkerby, 2016-07-09:
@travisgoodspeed tried again with a lab PSU (and thanks for the codeplug tip). What I've just noticed is that since I ran the modified firmware, the radio thinks it's VHF when I read the codeplug out whereas it's a UHF unit. See https://www.dropbox.com/s/72s6q13lz8hy51m/Screenshot%202016-07-09%2018.17.27.png?dl=0  Might not be this incompatability but there is definitely something going on. Retevis have offered to replace in case it's a hardware fault - is there anything useful for you in case we see similar faults before I send it back?

eb4fbz, 2016-07-10:
Here http://blog.retevis.com/firmware-update-rt3-rt8/ Retevis says there is a difference in Vocoder. Also mentions about crashes with new LCDs.

travisgoodspeed, 2016-07-10:
According to the blog, it's a vocodee issue.  The old firmware emulated a vocoder in software; maybe the new one reverted to hardware?

I now have two GPS units on the way. As soon as one arrives, I'll start playing with this.

aeickho, 2016-07-10:
New vocoder, new  lizenz ("new #44 ")  ...

eb4fbz, 2016-07-10:
ALPU-MP IC is still present, IC marking is the same (1617). Maybe they have changed the key???

travisgoodspeed, 2016-07-10:
So at startup, the MD380 calls an AES encryption function twice.  The contents vary by radio, but are always the same for one radio.

```
iMac% md380-tool dmesg
17:14:21
Channel 2 selected.
Starting main()
aes_cipher(0x200125e4);
IN  : 20 68 91 d0 5a 00 2e 00 03 51 35 30 35 37 35 32
OUT : 9d 97 ab 63 fe 36 bc 5c fe c6 f4 4f 29 47 7e 12
aes_cipher(0x200125e4);
IN  : 20 68 91 d0 5a 00 2e 00 03 51 35 30 35 37 35 32
OUT : 8c db 08 94 f4 e0 cf 90 d7 a8 ce 2f ab 87 70 37

iMac%
```

When I apply the following patch to neuter the call to `aes_cipher()` from its hook, the screen boots to a blank white screen, much like when firmware is loaded to the wrong radio.  (It does not reboot, just hangs on a white screen.  The USB stack is fully functional, even if things like channel selection are unresponsive.)

```
iMac% git diff
diff --git a/applet/src/aes.c b/applet/src/aes.c
index 7a4475f..dbe4c78 100644
--- a/applet/src/aes.c
+++ b/applet/src/aes.c
@@ -129,8 +129,9 @@ char *aes_cipher_hook(char *pkt){
      use the original Tytera algorithm.  At some point, it might make
      sense to replace this with proper crypto, rather than XOR.
   */
+  printf("Skipping encryption.\n");
   printf("\nOUT :");
-  res=aes_cipher(pkt);
+  //res=aes_cipher(pkt);
   printhex(res,16);       //Print the keystream it produces. (First 49 bits are XOR'ed with the audio.)
   printf("\n");
   return res;
iMac% 
```

A partially redacted datasheet for the ALPU-MP is available at http://www.qsl.net/kb9mwr/projects/dv/dmr/ALPU-MP_datasheet_v2.16.pdf

dl8bb, 2016-07-10:
What's the meaning of the binary string '20 68 91 d0 5a 00 2e 00 03 51 35 30 35 37 35 32'. In ASCII it looks like this: "hÐZ.Q505752"

aeickho, 2016-07-10:
"h�ÐZ.�Q505752" see #26

travisgoodspeed, 2016-07-10:
False lead.  13.009 contains the same encryption keys.

dl8bb, 2016-07-10:
Hmmm, is '20 68 91 d0 5a 00 2e 00 03 51 35 30 35 37 35 32' really a key? IMHO not... it could be a good known string to verify that the right key is working...

travisgoodspeed, 2016-07-10:
No, those are data blocks.  Here's a trace with the keys and data.

```
iMac% md380-tool dmesg
2:3:18
Channel 1 selected.
Starting main()
aes_loadkey (0x20012604):  f5 c2 f1 8f 63 52 f5 a6 f0 6a a5 ed 5e 9c df 14
aes_cipher(0x200125e4);
IN  : 09 d2 ea 62 31 00 1d 00 01 51 35 30 35 37 35 32
OUT : a0 80 27 d5 db fd 49 57 fe c4 8a 7d cf a2 a0 9f
aes_loadkey (0x20012604):  dc 85 3a fa 51 f8 ae 79 7f a7 32 7b 15 59 03 95
aes_cipher(0x200125e4);
IN  : 09 d2 ea 62 31 00 1d 00 01 51 35 30 35 37 35 32
OUT : 8d b5 22 10 17 7d eb 0e bf b5 6f d0 4a b6 29 de

iMac% 
```

travisgoodspeed, 2016-07-10:
Created #168 for future discussion of the ALPU-MP chip, including what's known so far and example code.

M0IVQ, 2016-07-11:
The latest firmware information as posted in the Yahoo group:

S00x.xxx  S01x.xxx is for MD-380GPS and MD390GPS.
D00x.xxx  D01x.xxx is for MD-380 with out GPS and MD-390 with out GPS.

if correct then we have more than one issue to deal with as would appear that as well as the difference in vocoder, there is a difference in the firmware for GPS & non-GPS models.

Can anyone confirm if GPS models firmware begins with 'S' rather than 'D'?

DL9LBG, 2016-07-11:
Indeed the newest Retevis RT8 GPS Firmware is S013.014 .

d235j, 2016-07-11:
Huh. Is there a D013.014 as well? And if you flash S firmware on a radio with D firmware, does it stay S or D?

relocker, 2016-07-11:
A few weeks ago I tried upgating my TYT MD390 GPS with the the boot-loop as result. @^&#**##%(#
Finaly (thanks  M0IVQ for your hint) I downloaded the S-firmware AND IT WORKED.
I can play withh DMR again!

I found a working Firmware DM-390 GPS S013.012 975.50 KB (MCU version:	N011.006)
http://portofoontoebehoren.nl/product/tyt-portofoons/tyt-md-390-dmr-uhf-gps-portofoon/

d235j, 2016-07-12:
We've added the S firmware; it appears there are three firmware "forks" at present:
D00x.xxx: older MD-380 radios
D01x.xxx: newer radios without GPS
S01x.xxx: newer radios with GPS

If anyone locates firmware versions (even old ones) that aren't in https://github.com/travisgoodspeed/md380tools/blob/master/firmware/firmware_files.txt, please open a new issue.

DO9ULI, 2016-07-13:
New Vocoder-Firmware RT8-GPS
S013.014
[Retevis-Vocoder-RT8-S13.14.bin.txt](https://github.com/travisgoodspeed/md380tools/files/361250/Retevis-Vocoder-RT8-S13.14.bin.txt)

73  Uli DO9ULI

geiraage, 2016-07-13:
I got all new FW from TTY today. 

Look at this note 

Note:

1、MD-380 has two versions, new vocoder and old vocoder, so please check version first in
the radio, check menu-utilities- radio info-versions and if you find the version is D002 and
D003, it should the old vocoder and the latest firmware is MD380-D3.20, while if the
version is D013, the firmware should be MD380-D13.20

2、MD-446 has two versions, new vocoder and old vocoder, so please check version first in
the radio, check menu-utilities- radio info-versions and if you find the version is D002 and
D003, it should the old vocoder and the latest firmware is MD446-D3.20, while if the
version is D013, the firmware should be MD446-D13.20

3、MD-390 has three firmware, old vocoder and new vocoder with GPS and new vocoder
without GPS, so please check version first in the radio, check menu-utilities- radio info-
versions and if you find the version is D002 and D003, it should the old vocoder and the
latest firmware is MD380-D3.20, while if the version is D013, and radio has GPS, the
firmware should be MD390-S13.20 and if radio doesn’t have GPS, it should be MD380-
D13.20

3、MD-380G has two firmware, new vocoder with GPS and new vocoder without GPS, so
please check version first in the radio, check menu-utilities- radio info-versions and if the
version is D013, and radio has GPS, the firmware should be MD390-S13.20 and if radio
doesn’t have GPS, it should be MD380-D13.20

The fw files are located on my google drive public folder. 

https://drive.google.com/folderview?id=0ByLb3jTWrfDdeHNKUUdMZm9DSEU&usp=sharing

d235j, 2016-07-13:
@geiraage Nice, thanks! Whom did you get these from? It would be nice to obtain MD-280 firmware as well.

geiraage, 2016-07-13:
I agree. Now we know the latest version and with what device that's work
with correct fw. I got it directly with Tytera. I have a dispute with a
faulty Tyt MD 380 VHF. No sound on digital but works fine with analog.
They suggested to update firmware with this just released fw. What you see
is what I got. :-)
13. jul. 2016 2.41 p.m.

sconklin, 2016-07-14:
Thanks, my radio has the new vocoder and the new firmware let me recover a working radio. The CDROM that came with the radio contains the old (incompatible) firmware, fwiw.

Giorgiofox, 2016-07-14:
Problem seem solved, my radio with new vocoder is working 100% with 2.032 patched firmware, thanks to Alex. Travis will merge this.

M0IVQ, 2016-07-14:
Does this create a separate patched firmware file for new radios or is it a 'one file fits all'?

aeickho, 2016-07-14:
is it a one file fits on all md380 radios

cswilsn2k, 2016-07-15:
Just out of curiosity, how do you build the firmware for D013.009? I have been reading documentation for hours but have not come across how to patch/build the firmware for other radio revisions. I see make all downloads the firmware, but not sure if its patching it?

aeickho, 2016-07-15:
patch to running experimental fw on "md380 new vocoder (D01x.XXX) radio" on mainline with 9a4999395719911fa80e14d619b335c025f5c1ef

aeickho, 2016-07-15:
"we understand the hardware differences well enough to back-port support into our fork of 2.032."
i hope the "change of alpu" is the once difference. 
I'll close this issue now... hopefully

closed 2016-07-15 by aeickho

N7DKL, 2016-07-31:
I need a "wee" bit more help. I have a new MD-390 with GPS and I tried to upload the hacked firmware which did not work (I suppose because it's a 390 and not a 380). Now I can't get it back. I have tried multiple times to reload the S13.12 and S13.20 and I am stuck in the "white screen reboot" loop. I am not sure it is actually loading the entire firmware. After I put the radio in upload mode (flashing red and green) and make sure I have all the dll's running I start the download. The blue bar starts moving, the LED goes solid red for about 1/2 the blue bar, then solid green for a bit then back to flashing for about the last 1/3 of the blue status bar. The final box that pops up in in Klingon so I can't read what it is trying to tell me.  My battery is topped off, so it's not that. 

Ideas?

travisgoodspeed, 2016-07-31:
Try an S03.xx firmware image.  Does that work?

Do you remember which version of firmware your radio arrived with?

Did you compile the patched firmware from source?  We've had support for all know MD390 models for the past two weeks.

N7DKL, 2016-07-31:
I don't recall the firmware version it came with (stupid me).  I just tried the newest firmware version directly from the Tytera website (MD390-S13.20) and again it failed and I get the warning box I can't read. Maybe it doesn't like Windows 10? The programming software has been running fine. 

I may not have used the right software of yours. I may need to search around a bit to figure out how to do that. I am SO not a programmer.

N7DKL, 2016-07-31:
I got it from the VA3XPR site. It was the v2 version from April. I will search for the S03.xx version now.

d235j, 2016-07-31:
@N7DKL all known Tytera-original firmware versions can be found at https://archive.org/download/TYTMD380FW2.

N7DKL, 2016-07-31:
d235j, Thank you! I was able to use the TYT-GPS-MD390-MD380-S003.012.bin file to work so now the radio is back up an running. 

I would really like to be able to scan the talkgroups but I guess I will need more coffee (or wine) to get the experimental firmware to work.

travisgoodspeed, 2016-07-31:
Excellent!  Leaving this issue closed.

@d235j will be figuring out a nightly build process to give VA3XPR and others a more timely source for binaries. It will be tracked as a separate issue.