travisgoodspeed/md380tools #156: Firmware Download Bug, Silent Failure

travisgoodspeed, 2016-06-22:
Since #153, the Makefile silently fails if a download is the wrong size according to `du`.

Close this issue when:
- [ ] Download failures cause a make failure, instead of just quiet file deletion.
- [ ] Check filesize with `ls -l` instead of `du`, which varies by the file size.

I'll assign @d235j when he accepts his damned commit rights.

ajorg, 2016-06-22:
Accidentally posted this to the pull-request discussion instead of here:
> du is meant to tell you how much of your disk the file is actually using, it's not meant to tell you the size of the file. stat -c %s is a better way to do that.

ajorg, 2016-06-22:
Should also use curl's --fail option to make it give you a non-zero exit code on failure.

d235j, 2016-06-22:
@ajorg yeah, as a first pass I didn't do much cleanup of the already-existing download code. Among other things I dislike the hardcoded list in the Makefile — I'd rather pull the information from a separate text file.

@ajorg: stat -c %s is GNU behaviour and does not work on Mac OS X; I will investigate a better solution.

d235j, 2016-06-22:
@ajorg any objections to using ```wc -c <file```?

ajorg, 2016-06-22:
@d235j not for files this small. for larger files I'd object strenuously.

d235j, 2016-06-22:
@ajorg good point, not sure how I missed that.. Think I'll use ```find filename -a -size Nc``` instead.

d235j, 2016-06-22:
Should be fixed as of c795189. I will await testing by others before closing this issue.

d235j, 2016-06-23:
As of  e076985, the Makefile is mostly generic and reads firmware information from a separate text file. Additionally, both the filesize and sha256 hashes are checked.

d235j, 2016-06-23:
Closing -- please comment or open a new issue if there are further problems.

closed 2016-06-23 by d235j