travisgoodspeed/md380tools #160: GPS Hardware

travisgoodspeed, 2016-07-07:
While #155 is to get support for booting on units with GPS support, this issue is to actually use the GPS hardware, which is a Jumpstar JS-M710 module.
http://en.defenstar.com/products_detail/&productId=44.html

A photo from atg in the IRC channel shows the module.  We don't yet have a pinout.
[image]

Close this issue when GPS is supported.

eb4fbz, 2016-07-08:
Module is Defenstar JS-M710 http://en.defenstar.com/products_detail/&productId=44.html No datasheet available.
I think GPS UART is at ZIF connector pins 19 and 20. Connectors and flex cable have been increased from 24 to 26pins to accomodate this. 
Main board should be different too in GPS units, MD390 and RT8 without GPS have 24pin connectors.

travisgoodspeed, 2016-07-08:
Hopefully the differences aren't major.  If at all possible, I'd like to hook a few low-level I/O functions and produce an image that's compatible with both forks of the hardware.

f4nen, 2016-07-08:
There is a product specification online with a pinout for at least the GPS module:
http://www.docin.com/p-888414669.html
Cheers!

d235j, 2016-07-09:
Here's the PDF.
[JS-M710.pdf](https://github.com/travisgoodspeed/md380tools/files/355732/JS-M710.pdf)

zarya, 2016-08-28:
GPS dataframe from the MD390 as received by BM

Data Header (12 bytes):  0d 05 03 20 c3 1f 2c a4 00 1b c1 4e  --  ......,....N
Data Continuation (12 bytes):  70 0d 08 84 28 15 54 41 80 18 66 2a  --  p...(.TA..f*
Data Continuation (12 bytes):  70 0d 08 84 28 15 54 41 80 18 66 2a  --  p...(.TA..f*

eb4fbz, 2016-09-04:
GPS module TXD is connected to MCU USART3_RX (pin 56). GPS RXD is not connected, although the signal is present in the FPC connector.

Protocol is NMEA at 9600bauds: GPGGA, GPGLL, GPGSA, GPGSV, GPRMC and GPVTG.

aeickho, 2016-09-04:
Thanks this is a nice info... 

We close this issue when we are "understanding" the gps part ;)

travisgoodspeed, 2016-09-04:
Do we understand why D13 won't boot on GPS devices and S13 won't boot on non-GPS devices yet?  What was Pin 56 used for non non-GPS devices?

eb4fbz, 2016-09-04:
According to the published schematics (known to not fully match with md380), pin 55 and 56 are PLL_LD and PLL_CS (PLL SPI bus). In MD390 with GPS they added two 100K pull-up resistors.

eb4fbz, 2016-09-12:
Hi all. Given no one got a clue about the GPS data format, i decided to inject simulated NMEA frames into my MD390 and dump the decoded DMR frames with DSD. With help of a friend (EA4EOZ) we almost got all the information needed. Will post soon!

eb4fbz, 2016-09-14:
Just to close this issue:

bit 2 : 1=N, 0=S
bit 3 : 1=E, 0=W
bit 4 : Fix
bits(5:11): Speed in knots/2
bits(12:18): Latitude. Degrees
bits(19:24): Latitude. Minutes integer part
bits(25:38): Latitude. Minutes decimal part
bits(39:46): Longitude. Degrees
bits(47:52): Longitude. Minutes integer part
bits(53:66): Longitude. Minutes decimal part
bits(67:80): Altitude in meters, only least 8bits used (bug)
bits(81:96): Garbage?

Only data from $GPRMC is used. Course is not encoded and altitude is buggy: just 8bits are used so no negative or higher than 255m elevations are allowed.

trolfe, 2016-09-17:
Thanks for the format info - I have done some coding and I think the speed knots/2 is more likely to be speed m/s? Also to clarify the Min decimal part are decimals x 10000? Great work and really useful info.

eb4fbz, 2016-09-20:
It's knots/2 (knots >> 1). I've reversed it using fake NMEA GPRMC frames.
About decimal part, i will check my notes in a few days, i'm now traveling in China :(

travisgoodspeed, 2016-12-23:
Closing issue as S13 is stable and we can continue GPS reversing from there.

closed 2016-12-23 by travisgoodspeed