travisgoodspeed/md380tools #162: Feature: End-of-call chirp

nivex, 2016-07-09:
When a received call ends, play a short chirp on the speaker as an audible cue to the user who might not be looking at the handset (eg: using a speaker mic).

Some folk I've talked to in the area miss the "courtesy tone" of the local analog repeaters. This would be a way to replicate that functionality.

aeickho, 2016-07-09:
please tray
    Radio Settings->Tones/Alerts->All Tones -> Turn On

i think this solved your wish.

travisgoodspeed, 2016-07-09:
@aeickho Does this give an ending tone for you?  On my radios, it only gives the initial chirp, not one after the end of the broadcast.

aeickho, 2016-07-09:
@travisgoodspeed 
I gave you a short movie via twitter sent with respect to the behavior.

I'm unsure if we're talking about the same.

good night from Germany

travisgoodspeed, 2016-07-09:
My mistake; this feature does already exist.  Closing the issue.

closed 2016-07-09 by travisgoodspeed

nivex, 2016-07-25:
I just got a chance to flash the most current firmware this morning. I have verified that All Tones is turned on. I do not hear any tone when the far end ceases their transmission.

aeickho, 2016-07-25:
I'll will check it... Mybee an Problem included by the last chage re #130?

aeickho, 2016-07-25:
No ... It works fine on my Radios with head of Master

aeickho, 2016-07-25:
Is there an misunderstanding?

nivex, 2016-09-21:
For reference, this needs the "Ch Free Indication Tone" checkbox set in the CPS.