travisgoodspeed/md380tools #163: Feature: Send ID digitally

nivex, 2016-07-09:
Might it be possible to send an ID digitally as part of each call?

I could see either using an unused burst type or a text message to a designated ID to contain the information. Using the contents of Radio Name would be an easy way to store the callsign without needing extra bits.

It would be nice to be able to comply with the FCC identification requirements digitally instead of speaking the callsign every 10 minutes.

Bonus would be if such a burst would make it through a repeater and other modified MD380s could receive it, alleviating the need for the full database in the radio.

travisgoodspeed, 2016-09-23:
Merged into #370. Reopen only if there is active progress in adding such a feature.

closed 2016-09-23 by travisgoodspeed

sijskes, 2016-10-01:
only the lastest version of the standard has 'talker alias'. so i dont think it will be implemented soon in other handsets.

travisgoodspeed, 2016-10-01:
I really like this feature on Yaesu System Fusion radios, but our Caller ID Database is pretty damned close in functionality.

Reopen this issue if/when Motorola or Hytera ships a radio that actually uses the Talker Alias parts of the protocol.  We can then make one that's compatible.

sijskes, 2016-10-01:
https://portal.etsi.org/ngppapp/ContributionSearchForm.aspx?tbid=&SubTB=&Param=&MeetingId=16029

it was a change request from Hytera. CR028 so, a firmware update might include this in hytera's. 
i cannot promise the C5000 is transparent enough.

travisgoodspeed, 2016-11-30:
PD0ZRY helpfully mentions that the feature is documented in the ETSI spec.
https://wiki.brandmeister.network/index.php/Talker_Alias

I still have no local Brandmeister repeaters, but I've bought a few Hytera radios that might be able to test for compatibility of this standard.  If I get them up and running with it, then I might reopen the issue.

Does anyone know if Motorola intends to implement this portion of the standard?

73 from NYC,
--KK4VCZ