travisgoodspeed/md380tools #164: make flash - failed?

dg9vh, 2016-07-09:
After doing git pull to get actual copy of repo trying to do a make flash ends up in a timeout:
```
kim@dg9vh-hp:~/src/md380tools$ make flash
cd applet && make flash
make[1]: Entering directory '/home/kim/src/md380tools/applet'
make[1]: Zirkuläre Datei src/md380-2.032.c <- src/version.h Abhängigkeit wird nicht verwendet.
make -C lib
make[2]: Entering directory '/home/kim/src/md380tools/applet/lib'
arm-none-eabi-ar -r libstm32f4.a misc.o stm32f4xx_dma.o stm32f4xx_rcc.o stm32f4xx_adc.o stm32f4xx_exti.o stm32f4xx_rng.o stm32f4xx_can.o stm32f4xx_flash.o stm32f4xx_rtc.o stm32f4xx_crc.o stm32f4xx_fsmc.o stm32f4xx_sdio.o stm32f4xx_cryp_aes.o stm32f4xx_gpio.o stm32f4xx_spi.o stm32f4xx_cryp.o stm32f4xx_hash.o stm32f4xx_syscfg.o stm32f4xx_cryp_des.o stm32f4xx_hash_md5.o stm32f4xx_tim.o stm32f4xx_cryp_tdes.o stm32f4xx_hash_sha1.o stm32f4xx_usart.o stm32f4xx_dac.o stm32f4xx_i2c.o stm32f4xx_wwdg.o stm32f4xx_dbgmcu.o stm32f4xx_iwdg.o stm32f4xx_dcmi.o stm32f4xx_pwr.o
make[2]: Leaving directory '/home/kim/src/md380tools/applet/lib'
# use an fresh experiment.img
cp ../patches/2.032/patched.img ./experiment.img
python2 merge.py experiment.img main.img 0x0809D000
Merging an applet.
Loading symbols from main.img.sym
Hooking a menu call.
Merging main.img into experiment.img at 0809d000
../md380-fw --wrap experiment.img experiment.bin
DEBUG: reading "experiment.img"
INFO: base address 0x800c000
INFO: length 0xf2c00
DEBUG: writing "experiment.bin"
../md380-dfu upgrade experiment.bin
[Errno 110] Operation timed out
make[1]: Leaving directory '/home/kim/src/md380tools/applet'
```

while a make flashdb for updating the usersdb works perfectly:
```
kim@dg9vh-hp:~/src/md380tools$ make flashdb
cd db && make
make[1]: Entering directory '/home/kim/src/md380tools/db'
curl 'http://www.dmr-marc.net/cgi-bin/trbo-database/datadump.cgi?table=users&format=csv&header=0' | perl -pe 's,<br/>,,' >users.csv
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 2832k    0 2832k    0     0  47883      0 --:--:--  0:01:00 --:--:-- 40467
make[1]: Leaving directory '/home/kim/src/md380tools/db'
iconv -c -f UTF-8 -t ascii//TRANSLIT db/users.csv | cut -d',' -f1-3,5-6 | sed 's/,\s+/,/g' > data.csv
wc -c < data.csv > data
cat data.csv >> data
./md380-tool spiflashwrite data 0x100000
SPI Flash ID: ef 40 18
W25Q128FV 16MByte
erase 1912998 bytes @ 0x100000
flashing 1912998 bytes @ 0x100000
reboot radio now
```

Anyone else with that problem? No idea where the problem is.

DG8ZO, 2016-07-09:
Yes, i have not the exact similar error but its very close.
I tried to fash my Retevis RT3 with make clean flash after i updated my local git copy with git pull. The RT3 is in Recovery Mode.
But i get these error:

Merging main.img into experiment.img at 0809d000
../md380-fw --wrap experiment.img experiment.bin
DEBUG: reading "experiment.img"
INFO: base address 0x800c000
INFO: length 0xf2c00
DEBUG: writing "experiment.bin"
../md380-dfu upgrade experiment.bin
The device does not support the specified langid
make[1]: Leaving directory '/home/dg8zo/Downloads/md380tools/applet'

travisgoodspeed, 2016-07-09:
What operating system and language localization are the two of you running?

I suspect that f6451bbdb1ba7f16956d47d595dfb37009d869c0 might be to blame.  The purpose of that patch was to verify that the device is in its bootloader mode, rather than its application mode.  Do you experience the same problems when you revert that patch?

DG8ZO, 2016-07-09:
I'm on (L)Ubuntu 15.10 with German Localization.

I try to revert the patch yet. Please give me one Moment ;-)

dg9vh, 2016-07-09:
@travisgoodspeed I am using Debian 8 (64Bit)
`Linux dg9vh-hp 3.16.0-4-amd64 #1 SMP Debian 3.16.7-ckt25-2 (2016-04-08) x86_64 GNU/Linux`

DG8ZO, 2016-07-09:
@travisgoodspeed 
Yes you are right Travis, revert the patch and alll works!

d235j, 2016-07-09:
@travisgoodspeed is it even necessary to pass the langid here?

DG8ZO, 2016-07-09:
@dg9vh 
Can you test the same?
I commented out the lines in the corresponding file with a # by hand and then
i make a "make clean flash". It works for me.

dg9vh, 2016-07-09:
@DG8ZO I reverted the named patch and it made me flashing again...

travisgoodspeed, 2016-07-09:
I'm writing a proper fix now, will push a patch within the hour.

DG8ZO, 2016-07-09:
@travisgoodspeed 
Ok, perfect. It's 10:30PM in Germany but i stay awake to test it,

dg9vh, 2016-07-09:
@DG8ZO I'll be there, too, at least till 23:30 MESZ (till "Independence Day" is over in TV)... so testing would be suitable.

travisgoodspeed, 2016-07-09:
@DG8ZO and @dg9vh, please check to see if that patch worked for you in Linux.

It breaks support for OS X, but I'll have that finished soon.

DG8ZO, 2016-07-09:
@travisgoodspeed 
Ok, have tested and here it works perfectly!

dg9vh, 2016-07-09:
@travisgoodspeed here too

travisgoodspeed, 2016-07-09:
Reopening until a patch for Mac is ready.

The root cause is that usb.util.get_string() takes three parameters on Linux and four parameters on Mac.  Yay consistency!  I'll close this issue when my patch is merged and tested.

dg9vh, 2016-07-09:
@travisgoodspeed ok - do it :+1:

travisgoodspeed, 2016-07-09:
Works on both Mac and Linux again.  Closing the issue.

closed 2016-07-09 by travisgoodspeed

d235j, 2016-07-09:
@travisgoodspeed are you sure that you have an up-to-date PyUSB? It seems they removed the length argument in https://github.com/walac/pyusb/commit/dac78933f6a6eaf5ae82f48e2f4e7d1733dc2f98, and the version on my disk only takes three arguments too.

travisgoodspeed, 2016-07-09:
I see the old calling convention in the version of pyusb that ships with recent MacPorts, and the new calling convention in Debian/Stretch.

d235j, 2016-07-09:
Great, looks like MacPorts still ships a really old version of pyusb; they have a newer version as py-pyusb-devel, but even this is 1.0.0b2 while 1.0.0 was released in April.

Cabalist, 2017-01-26:
Hello,

I am on macOS 10.12.2 (16C67) with libusb: stable 1.0.21, and pyusb (1.0.0) and python 2.7.10

I am getting errors very similar to the ones discussed here.  I have tried flashing multiple times, I am running the latest commit (50b7fba)

I get the following errors without consistency: 

```
./md380-dfu upgrade applet/experiment.bin
Beginning firmware upgrade.
[Errno 5] Input/Output Error
```

```
./md380-dfu upgrade applet/experiment.bin
Beginning firmware upgrade.
[Errno 60] Operation timed out
```

```
./md380-dfu upgrade applet/experiment.bin
The device has no langid
```

```
./md380-dfu upgrade applet/experiment.bin
[Errno 84] Overflow
make: *** [flash] Error 1
```
I have looked at the traceback and it seems to be related to LibUSB.  I am not sure what the next troubleshooting steps should be.  Below are the tracebacks when I tried to flash the radio using the following steps:

1. Plug in USB
2. Plug in Radio
3. Turn on radio while holding down PTT and button about PTT
4. ```make flash```
5. If it fails run it again.

```
./md380-dfu upgrade applet/experiment.bin
Beginning firmware upgrade.
[Errno 5] Input/Output Error
Traceback (most recent call last):
  File "/Users/work/PycharmProjects/md380tools/md380_dfu.py", line 251, in download_firmware
    dfu.erase_block(address)
  File "/Users/work/PycharmProjects/md380tools/DFU.py", line 188, in erase_block
    self.enter_dfu_mode()
  File "/Users/work/PycharmProjects/md380tools/DFU.py", line 286, in enter_dfu_mode
    action()
  File "/Users/work/PycharmProjects/md380tools/DFU.py", line 256, in abort
    self._device.ctrl_transfer(0x21, Request.ABORT, 0, 0, None)
  File "/Users/work/.virtualenvs/md380tools27/lib/python2.7/site-packages/usb/core.py", line 1043, in ctrl_transfer
    self.__get_timeout(timeout))
  File "/Users/work/.virtualenvs/md380tools27/lib/python2.7/site-packages/usb/backend/libusb1.py", line 883, in ctrl_transfer
    timeout))
  File "/Users/work/.virtualenvs/md380tools27/lib/python2.7/site-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
USBError: [Errno 5] Input/Output Error
```

```
./md380-dfu upgrade applet/experiment.bin
Beginning firmware upgrade.
[Errno 60] Operation timed out
Traceback (most recent call last):
  File "/Users/work/PycharmProjects/md380tools/md380_dfu.py", line 246, in download_firmware
    dfu.md380_custom(0x91, 0x31)
  File "/Users/work/PycharmProjects/md380tools/DFU.py", line 206, in md380_custom
    self.enter_dfu_mode()
  File "/Users/work/PycharmProjects/md380tools/DFU.py", line 282, in enter_dfu_mode
    state = self.get_state()
  File "/Users/work/PycharmProjects/md380tools/DFU.py", line 252, in get_state
    state_packed = self._device.ctrl_transfer(0xA1, Request.GETSTATE, 0, 0, 1)
  File "/Users/work/.virtualenvs/md380tools27/lib/python2.7/site-packages/usb/core.py", line 1043, in ctrl_transfer
    self.__get_timeout(timeout))
  File "/Users/work/.virtualenvs/md380tools27/lib/python2.7/site-packages/usb/backend/libusb1.py", line 883, in ctrl_transfer
    timeout))
  File "/Users/work/.virtualenvs/md380tools27/lib/python2.7/site-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
USBError: [Errno 60] Operation timed out
```

```
./md380-dfu upgrade applet/experiment.bin
Beginning firmware upgrade.
[Errno 5] Input/Output Error
Traceback (most recent call last):
  File "/Users/work/PycharmProjects/md380tools/md380_dfu.py", line 245, in download_firmware
    dfu.md380_custom(0x91, 0x01)
  File "/Users/work/PycharmProjects/md380tools/DFU.py", line 202, in md380_custom
    status = self.get_status()  # this gets the status
  File "/Users/work/PycharmProjects/md380tools/DFU.py", line 243, in get_status
    status_packed = self._device.ctrl_transfer(0xA1, Request.GETSTATUS, 0, 0, 6)
  File "/Users/work/.virtualenvs/md380tools27/lib/python2.7/site-packages/usb/core.py", line 1043, in ctrl_transfer
    self.__get_timeout(timeout))
  File "/Users/work/.virtualenvs/md380tools27/lib/python2.7/site-packages/usb/backend/libusb1.py", line 883, in ctrl_transfer
    timeout))
  File "/Users/work/.virtualenvs/md380tools27/lib/python2.7/site-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
USBError: [Errno 5] Input/Output Error
```

-----
Sidenote:  
As an interest to the debugging I added a loop that polled dfu.get_status() every 5 second.  It times out with irregularlity (sometimes after 2 attempts, sometimes after 10, etc) with the following traceback:

```
        for _ in range(100):
            print( dfu.get_status())
            time.sleep(5)
```

```
(OK, 0, dfuIDLE, 0)
(OK, 0, dfuIDLE, 0)
[Errno 60] Operation timed out
Traceback (most recent call last):
  File "/Users/work/PycharmProjects/md380tools/md380_dfu.py", line 246, in download_firmware
    print( dfu.get_status())
  File "/Users/work/PycharmProjects/md380tools/DFU.py", line 243, in get_status
    status_packed = self._device.ctrl_transfer(0xA1, Request.GETSTATUS, 0, 0, 6)
  File "/Users/work/.virtualenvs/md380tools27/lib/python2.7/site-packages/usb/core.py", line 1043, in ctrl_transfer
    self.__get_timeout(timeout))
  File "/Users/work/.virtualenvs/md380tools27/lib/python2.7/site-packages/usb/backend/libusb1.py", line 883, in ctrl_transfer
    timeout))
  File "/Users/work/.virtualenvs/md380tools27/lib/python2.7/site-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
USBError: [Errno 60] Operation timed out
``

claywalker0929, 2017-01-26:
Is your radio cable plugged in to a USB hub or into the computer? It sounds like you might be having noise issues.

Cabalist, 2017-01-26:
It is plugged into the side of my computer.

dg9vh, 2017-01-26:
@Cabalist what happens in /var/log/syslog when switching the radio on in normal mode?

Cabalist, 2017-01-26:
Turning on the radio without holding anything gives me this:

```
default	12:37:46.885754 -0800	icdd	#ICDebug - 23:{ICWiredBrowser.m} (USB Device first match)
default	12:37:46.886951 -0800	icdd	#ICDebug - 388:{ICWiredBrowser.m} (1 USB Descriptions Managed)
default	12:37:46.889088 -0800	icdd	#ICDebug - 460:{ICDDMessageCenter.m} (+Add Digital Radio in USB mode - 0x0/0x0/0x0 - 0x14100000 - ICDeviceDescriptionSUQuery)
default	12:37:46.937373 -0800	icdd	#ICDebug - 213:{ICResourceManager.m} (00000000-3030-3030-3030-303030313043|Digital Radio in USB mode|MANUFACTURER:AnyRoad Technology;MODEL:Digital Radio in USB mode|SW=FALSE|)
default	12:37:46.939065 -0800	icdd	#ICDebug - 460:{ICDDMessageCenter.m} (+Add Digital Radio in USB mode - 0x0/0x0/0x0 - 0x14100000 - ICDeviceDescriptionAdded)
default	12:37:46.975984 -0800	icdd	#ICDebug - 37:{ICWiredBrowser.m} (USB Interface first match)
default	12:37:46.977234 -0800	icdd	#ICDebug - 388:{ICWiredBrowser.m} (2 USB Descriptions Managed)
default	12:37:46.981346 -0800	icdd	#ICDebug - 460:{ICDDMessageCenter.m} (+Add Digital Radio in USB mode - 0xfe/0x1/0x2 - 0x14100000 - ICDeviceDescriptionSUQuery)
default	12:37:46.981422 -0800	icdd	#ICDebug - 213:{ICResourceManager.m} (00000000-0000-0000-0000-00000483DF11|Digital Radio in USB mode|(null)|SW=FALSE|)
default	12:37:46.984333 -0800	icdd	#ICDebug - 460:{ICDDMessageCenter.m} (+Add Digital Radio in USB mode - 0xfe/0x1/0x2 - 0x14100000 - ICDeviceDescriptionUndefined)
```

Cabalist, 2017-01-26:
It is probably worth mentioning that after I put it in DFU mode it hasn't left it even after toggling power and removing the battery.  I assumed that it was normal after reading through a number of posts on forums.  Once you go to DFU it need to be flashed with something...

ladams00, 2018-04-19:
@Cabalist, I'm having the same trouble as you with the radio not flashing (I get [Errno 5] Input/Output Error), and the device remains in DFU mode even after power cycling it.  I will continue to work on it.

commit 088db045 (2016-07-09): Fixes a mis-use of usb.util.get_string's parameters to repair usage on Linux.  Close #164.