travisgoodspeed/md380tools #168: ALPU-MP Support and Hooking

travisgoodspeed, 2016-07-10:
Forked from #155, the purpose of this issue is to handle the ALPU-MP copy protection chip, hooking it until we understand it.  Close this issue when we understand the usage of the chip, how to communicate with it, and how to disable or live without it.

Datasheet:
http://www.qsl.net/kb9mwr/projects/dv/dmr/ALPU-MP_datasheet_v2.16.pdf

Example code:
https://github.com/wondermedia/wm8850/blob/f6730ac89e5e74811e2f4e2de5c6635253744866/ANDROID_3.0.8/drivers/char/wmt-alpu/s_wmt_alpu.c

aeickho, 2016-07-10:
currently I write low level hooks to dump the i2c communication

aeickho, 2016-07-11:
......a lot of telegrams, but only at system start

gn

aeickho, 2016-07-11:
I think: this is the while(1) loop when dongle check is wrong
```
see:  Func@ 0x08046e00
         0x08046e70      09f0e6fe       bl F_239_Nice

/ (fcn) F_180_Nice 2
\            `==-> 0x08046eb0      fee7           b F_180_Nice
```

aeickho, 2016-07-11:
@ 0x08032a3c

d235j, 2016-07-11:
Here's a PPT on the ALPU.
[ALPU.ppt.zip](https://github.com/travisgoodspeed/md380tools/files/358079/ALPU.ppt.zip)

aeickho, 2016-07-11:
re D013.009 ... alpu crypt handling is changed.

Is the secret changed? temporary I can't  find the secret on 13.009

gn

travisgoodspeed, 2016-07-12:
The AES keys are `f5 c2 f1 8f 63 52 f5 a6 f0 6a a5 ed 5e 9c df 14` and `dc 85 3a fa 51 f8 ae 79 7f a7 32 7b 15 59 03 95`.

I find both keys in all firmware images, including D13.009.

```
iMac% for foo in *.img; (echo $foo && hexdump $foo | grep --color "f5 c2 f1 8f")
D002.022.img
0089fa0 8f 32 a3 36 f5 c2 f1 8f 63 52 f5 a6 f0 6a a5 ed
D002.025.img
0088a30 8f 32 a3 36 f5 c2 f1 8f 63 52 f5 a6 f0 6a a5 ed
D002.026.img
00889b0 8f 32 a3 36 f5 c2 f1 8f 63 52 f5 a6 f0 6a a5 ed
D002.030.img
0088bf0 8f 32 a3 36 f5 c2 f1 8f 63 52 f5 a6 f0 6a a5 ed
D002.032.img
0088c30 8f 32 a3 36 f5 c2 f1 8f 63 52 f5 a6 f0 6a a5 ed
D002.034.img
0088d70 8f 32 a3 36 f5 c2 f1 8f 63 52 f5 a6 f0 6a a5 ed
D003.008.img
0089050 f5 c2 f1 8f 63 52 f5 a6 f0 6a a5 ed 5e 9c df 14
D013.009.img
0088ad0 f5 c2 f1 8f 63 52 f5 a6 f0 6a a5 ed 5e 9c df 14
iMac% for foo in *.img; (echo $foo && hexdump $foo | grep --color "dc 85 3a fa")
D002.022.img
0089f80 7b 70 e2 86 dc 85 3a fa 51 f8 ae 79 7f a7 32 7b
D002.025.img
0088a10 7b 70 e2 86 dc 85 3a fa 51 f8 ae 79 7f a7 32 7b
D002.026.img
0088990 7b 70 e2 86 dc 85 3a fa 51 f8 ae 79 7f a7 32 7b
D002.030.img
0088bd0 7b 70 e2 86 dc 85 3a fa 51 f8 ae 79 7f a7 32 7b
D002.032.img
0088c10 7b 70 e2 86 dc 85 3a fa 51 f8 ae 79 7f a7 32 7b
D002.034.img
0088d50 7b 70 e2 86 dc 85 3a fa 51 f8 ae 79 7f a7 32 7b
D003.008.img
0089030 dc 85 3a fa 51 f8 ae 79 7f a7 32 7b 15 59 03 95
D013.009.img
0088ab0 dc 85 3a fa 51 f8 ae 79 7f a7 32 7b 15 59 03 95
iMac% 
```

travisgoodspeed, 2016-07-12:
@aeickho  This decompiled function might be easier to read than chasing the mess of conditional jumps in assembly.

```
int __fastcall sub_8046E00(_BYTE *a1)
{
  _BYTE *v1; // r4@1                                                                                                                                                                                                                            
  int result; // r0@1                                                                                                                                                                                                                           
  unsigned int v3; // r0@5                                                                                                                                                                                                                      

  v1 = a1;
  byte_2001D40B = sub_8050068();
  result = (unsigned __int8)byte_2001D40B;
  if ( byte_2001D40B )
  {
    byte_2001D40B = sub_8050BCC();
    result = (unsigned __int8)byte_2001D40B;
    if ( byte_2001D40B )
    {
      byte_2001D40B = sub_8050068();
      result = (unsigned __int8)byte_2001D40B;
      if ( byte_2001D40B )
      {
        byte_2001D40B = sub_8050068();
        result = (unsigned __int8)byte_2001D40B;
        if ( byte_2001D40B )
        {
          *v1 = 16;
          v3 = (v40002800 + (unsigned __int8)sub_804FE94()) & 7;
          if ( v3 != 1 )
          {
            if ( v3 >= 1 )
            {
              if ( v3 == 3 )
              {
                while ( 1 )
                  ;
              }
              if ( v3 < 3 )
              {
                while ( 1 )
                  ;
              }
              if ( v3 == 5 )
              {
                while ( 1 )
                  ;
              }
              if ( v3 < 5 )
                sub_8046C94();
              if ( v3 == 7 )
              {
                while ( 1 )
                  ;
              }
              if ( v3 < 7 )
              {
                while ( 1 )
                  ;
              }
            }
            while ( 1 )
              ;
          }
          while ( 1 )
            ;
        }
      }
    }
  }
  return result;
}
```

aeickho, 2016-07-12:
@travisgoodspeed What decompiler do you use?  radeco? 
The view is fantastic.

d235j, 2016-07-12:
@aeickho those results look like those from the IDA Pro ARM decompiler — which unfortunately is very expensive.

travisgoodspeed, 2016-07-13:
Yup, HexRays is hellishly expensive but worth every dime of the license costs.

Do we know whether the ALPU-MP chip varies with the audio codec?  It might explain why the authors were forced to break compatibility.

--Travis

eb4fbz, 2016-07-13:
The easy way is to swap two ALPU-MP chips between new and old vocoder radios.

aeickho, 2016-07-14:
on 13.020:
   with:
```
    patcher.nopout((0x08033f30+0x18));
    patcher.nopout((0x08033f30+0x1a));

    patcher.nopout((0x08033f30+0x2e));
    patcher.nopout((0x08033f30+0x30));

    patcher.nopout((0x08033f30+0x44));
    patcher.nopout((0x08033f30+0x46));

    patcher.nopout((0x08033f30+0x5a));
    patcher.nopout((0x08033f30+0x5c));

    patcher.nopout((0x08033f30+0x70));
    patcher.nopout((0x08033f30+0x72));
    
    patcher.nopout((0x08033f30+0x86));
    patcher.nopout((0x08033f30+0x88));

    patcher.nopout((0x0804915c+0x12));
    patcher.nopout((0x0804915c+0x14));

    patcher.nopout((0x080499e2+0x12));
    patcher.nopout((0x080499e2+0x14));

    patcher.nopout((0x08049ca8+0x10));
    patcher.nopout((0x08049ca8+0x12));

    patcher.nopout((0x0804a134+0x10));
    patcher.nopout((0x0804a134+0x12));

    patcher.nopout((0x0804abc0+0x10));
    patcher.nopout((0x0804abc0+0x12));

```
old vocoder md380 works

d235j, 2016-07-14:
Nice! Though I wonder if new vocoder produces higher quality audio. Is there any obvious reason as to why they changed it?
Do you have a new vocoder radio?

aeickho, 2016-07-14:
Now I have one ;)
i have now on an old md380 the software for the new md380 running

aeickho, 2016-07-14:
i have _no_ md380 for the original 13.020 fw

aeickho, 2016-07-14:
next step: patch the 2.032 for running on a new md380

d235j, 2016-07-14:
@aeickho I can confirm that your patch works on my MD380. I think a next step would be figuring out if GPS can be autodetected — the next radio I'm getting is an MD390 with GPS which most likely runs S01x.xxx firmware.

d235j, 2016-07-14:
@aeickho I don't think the patch allows the new vocoder to function, though... is it working for you?

aeickho, 2016-07-14:
I have no idea about the new vocoder function and changes.

according to my thinking : the patch on 13.020 makes a new vocoder on the md380old  to work.

According to my knowledge : the vocoder ingredient of fw .

d235j, 2016-07-14:
All right. I do wonder whether I just don't have enough power to hit the repeater — once I have a second radio, testing will be a lot easier.

travisgoodspeed, 2016-07-14:
My RT8 with the new vocoder and GPS arrived.  I'm busy this weekend but ought to have time for porting next week.

d235j, 2016-07-14:
@aeickho it seems patching out the call to the procedure at 0x08033f30 also allows it to boot, but I'm not sure if that breaks operation as I don't have another radio to test with yet.

aeickho, 2016-07-14:
with 70f888adab8583144495ab177d045411b61bb9c6 
the experimental fw runs on new md380

**## attention** ... flashdb is not workig

Giorgiofox, 2016-07-14:
I have a new (vocoder) model with a running patched firmware (from Alex). 
It work 100%.
Many time to flashdb, but it work!!

d235j, 2016-07-14:
@aeickho does flashdb only not work on new md380?

Giorgiofox, 2016-07-14:
@d235j it work on both model now!

tytmd380, 2016-07-15:
Unfortunately my newer version MD380 still bootloops with the lastest firmware. Im wondering if I am doing something wrong when trying to upgrade it. I have factory D013.009 firmware and followed this procedure to upgrade my radio under ubuntu:

git clone https://github.com/travisgoodspeed/md380tools.git
cd md380tools
sudo make clean

With radio in DFU mode:
sudo make all flash

When I power on the radio without the cable connected, the PoC||GTFO screen shows and then the radio starts boot looping.

Giorgiofox, 2016-07-15:
Try this

aeickho, 2016-07-15:
Dear tytmd380,
      This path is temporary not in the main branch.   
See me note post from yesterday evening. 

I'm pushing the patch mainline in some hours.

Sorry for the confusion.

Alex

aeickho, 2016-07-15:
patch is now on md380tools main branch 9a4999395719911fa80e14d619b335c025f5c1ef

closed 2016-07-15 by aeickho

tytmd380, 2016-07-16:
@aeickho Thanks for your efforts! Its working great now, I am new to github/code in general so I didn't realize that the latest fix wasn't included when I downloaded the code. Cheers