travisgoodspeed/md380tools #171: Simplex repeater function

kkomarica, 2016-07-11:
Not sure if this is possible, but the basic idea is to create a repeater mode which would allow the radio to receive on timeslot 1 and simultaneously re-transmit the same signal on timeslot 2 (same frequency, group, cc etc.). 

That way the coverage in areas without the repeater could be increased, or it could provide some cheap repeater solutions. 

Similar is function is available on tetra (DMO repeater). Some basic info on that is here: http://www.slideshare.net/zahidtg/tetra-direct-mode-operation

troffasky, 2016-07-15:
Don't assume this is possible to achieve in software alone. If the radio hardware is not full duplex then you can't receive and transmit simultaneously.

ajorg, 2016-07-15:
The DMR specs talk a lot about duplex communication over a single channel, including full duplex voice, but I've never heard of a radio that's capable of doing this, and it's hard to know if the limitation is in hardware or software.

d235j, 2016-07-15:
The HR_C5000 datasheet does suggest that it supports full duplex operation. Whether this is accurate is yet to be seen.

kkomarica, 2016-07-16:
I just have an idea, but unfortunately no technical knowledge that could help.
First step would be to confirm that the md380/390 can in fact receive and transmit at the same time. That would also help in maybe having a full duplex comm function in the future as it's closely link to this. 

If that is possible then we should think of the best way to implement the repeater idea. What comes to mind is that radios used in that system should be able to select TS 1 for RX and TS 2 for TX, and the repeater radio should be set vice versa. But I'm sure that the people here have much better ideas :)

troffasky, 2016-07-16:
Getting in to speculative territory here now, but I reckon if MD380 shipped with full duplex capability then it would be promoted as such rather than hidden away.

travisgoodspeed, 2016-10-01:
Merged into #370. Reopen only if there is active progress in adding such a feature.

closed 2016-10-01 by travisgoodspeed