travisgoodspeed/md380tools #173: Facts about the new vocoder used by retevis

aeickho, 2016-07-12:
on 
   http://blog.retevis.com/firmware-update-rt3-rt8/ 
is an note about a  "new Vocoder".

Let us collect here what changes in the vocoder have been made.

Close this Issue when the "improvement"  is documented :laughing:

d235j, 2016-07-12:
I just added an S003.012.bin firmware for older GPS radios in 4bc9727c5fc118047d0b55c468441107176b60c3. A vbindiff between this firmware and the S013.012 firmware shows that they are very similar; I'd probably poke at these to see what they changed with the new "vocoder".

travisgoodspeed, 2016-07-12:
With the new symbol migration tool, we can see that S013.012 does lack the ambe_encode_thing() and ambe_decode_wav() functions that are easily identified in S003.013.  They do use different vocoders!

S03.012 (old vocoder)
```
/* Unknown AMBE2+ thing. */
ambe_encode_thing    = 0x08050d91; /* 1024 byte match */
/* Decodes an AMBE2+ frame into bits. */
ambe_unpack          = 0x08048c9d; /* 38 byte match */
/* Populates the audio buffer */
ambe_decode_wav      = 0x08051249; /* 504 byte match */
```

S13.012 (new vocoder)
```
/* Unknown AMBE2+ thing. */
/* ambe_encode_thing has bad match of 4 points at 0x08050d91*/
/* Decodes an AMBE2+ frame into bits. */
ambe_unpack          = 0x08048c9d; /* 38 byte match */
/* Populates the audio buffer */
/* ambe_decode_wav has bad match of 2 points at 0x08051249*/
```

You can try the tool yourself with `make clean all` in `symbols/`.  All matches are made against the 2.032 firmware.

travisgoodspeed, 2016-07-12:
Updated my last comment now that the tool is printing the destination addresses.  Before it was printing the source addresses, which were quite a bit less useful.

adityanag, 2016-07-16:
I have this new radio and am willing to beta test the firmware updates if you like. I have the original firmware so I can recover if I need to.

adityanag, 2016-07-16:
Update - I have been able to successfully install the firmware on my radio with the new vocoder. Everything is working fine.

travisgoodspeed, 2016-07-28:
@aeickho Shall we close this issue, now that we have compatibility of each vocoder on the other's hardware?

aeickho, 2016-07-28:
my related question was ... what is difference between the too vocoder ... 

ok i  think the question is to "complicated" ... I'll close

closed 2016-07-28 by aeickho