travisgoodspeed/md380tools #177: libnewlib-arm-none-eabi is needed for compilation

corsac-s, 2016-07-15:
On Debian Sid it seems that the package  libnewlib-arm-none-eabi is needed to build the firmware, or you end up with:

```
make[2]: Leaving directory '/home/corsac/projets/sdr/md380/md380tools/applet/lib'
arm-none-eabi-gcc -std=gnu99 -g -O2 -Wall -Tmd380_applet.ld -mlittle-endian -mthumb -mcpu=cortex-m4 -mthumb-interwork -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fshort-wchar -Iinc -Ilib -Ilib/inc -I. -Ilib/inc/core -Ilib/inc/peripherals   -c -o main.o src/main.c
In file included from lib/inc/core/core_cm4.h:143:0,
                 from lib/inc/stm32f4xx.h:245,
                 from lib/inc/stm32f4_discovery.h:32,
                 from src/main.c:7:
/usr/lib/gcc/arm-none-eabi/4.9.3/include/stdint.h:9:26: fatal error: stdint.h: No such file or directory
 # include_next <stdint.h>
                          ^
compilation terminated.
<builtin>: recipe for target 'main.o' failed
make[1]: *** [main.o] Error 1
make[1]: Leaving directory '/home/corsac/projets/sdr/md380/md380tools/applet'
Makefile:16: recipe for target 'applets' failed
make: *** [applets] Error 2
```

Minor issue (obviously), the package could be added to the README.md (I can make a PR if you want but the issue seems so trivial I'm not sure it's worth it)

travisgoodspeed, 2016-07-15:
Yep, that's an omission. Close this issue when both the English and German docs have been fixed.

closed 2016-07-21 by aeickho

underdog, 2017-04-01:
So,
I'm still seeing this error in my centos 7 box.  I have looked for libnewlib libraries and not seeing any in yum database for centos.  I've downloaded the source code for newlib from here, https://sourceware.org/newlib/  I've complied it and installed it and still no joy

arm-none-eabi-gcc -std=gnu99 -g -O2 -Wall -Tapplet_d13.020.ld -mlittle-endian -mthumb -mcpu=cortex-m4 -mthumb-interwork -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fshort-wchar -Wa,-a=main.lst -DFW_D13_020	 -Iinc -Ilib -Ilib/inc -I. -Ilib/inc/core -Ilib/inc/peripherals -D "MD380_d13_020"     -c -o main.o src/main.c
In file included from lib/inc/core/core_cm4.h:143:0,
                 from lib/inc/stm32f4xx.h:245,
                 from lib/inc/stm32f4_discovery.h:32,
                 from src/main.c:7:
/usr/lib/gcc/arm-linux-gnueabi/4.8.1/include/stdint.h:9:26: fatal error: stdint.h: No such file or directory
 # include_next <stdint.h>
                          ^
compilation terminated.
make[1]: *** [main.o] Error 1

mbramwel, 2017-07-06:
any updates on this for Centos 7?

I installed the EL7 gcc-arm-linux-gnu from the Centos 7 repo
I then did a softlink:    ln -s arm-linux-gnu-gcc arm-none-eabi-gcc
"make flash" starts to process everything and eventually stops at the exact same spot as the above poster.

I checked and that file exists, not sure what <stdint.h> is being referenced here.

travisgoodspeed, 2017-07-06:
`arm-linux-gnu-gcc` and `arm-none-eabi-gcc` are different compilers targeting different platforms.  Remove your symlinks and install the correct compilers.

```
# dnf install arm-none-eabi-newlib gcc-arm-none-eabi
```

mbramwel, 2017-07-07:
Thanks for the response.  I have un-installed the previous packages as recommended.  Those 2 packages that you list are not part of any of the Standard Centos 7 repos (unless I am missing something somewhere).  I had to go to an older set of Fedora 19 packages otherwise obscene number of dependencies issues occur. The firmware tries to compile and eventually I get the following error:

 

arm-none-eabi-gcc: fatal error: selected multilib '.' not installed

 

I have googled the error but nothing useful was found. Yes, arm-none-eabi-newlib is installed.

 

FYI:  This all worked perfectly under Ubuntu, not going well under Centos 7 Server.  Have you tried Centos 7 Server?  If yes, what RPMs did you use?  There seem to be many versions on the Internet with the same name but are compiled against different glibc  (which is causing the dependency issues if I do not use fc19 rpms)

mbramwel, 2017-07-08:
I managed to get a working setup with Centos 7.3
I have documented the process to help others.

The process is here:    [http://web1.foxhollow.ca/?menu=centos7arm](url)

thanks guys, keep the updates coming!