travisgoodspeed/md380tools #178: Post-firmware install screen flipped and backwards

iball, 2016-07-16:
Pulled down the latest md380tools as of 16JUL16 @ 1227hrs CST and flashed my MD-390 that does NOT have a GPS.  
Now the screen is flipped upside down and the text is all backwards.  Everything works just fine except for that, even loading the DMR ID database (i.e. users.csv).
I'm kind of laughing about the whole thing myself.  So strange.
[image]

iball, 2016-07-16:
Additional notes:  Upgrade was performed on a brand-new MD-390 that does not have a GPS module.  
Upgrade was performed using latest md380tools on a fully-updated Raspberry Pi 3.
Even flashed the latest from https://pd0zry.nl/md380-fw/ just to see if my development environment was somehow screwed up but while it flashes just fine, it exhibits the same exact behavior of a flipped and upside-down screen.

travisgoodspeed, 2016-07-16:
Howdy!

Which version of firmware did the radio have when it arrived?  Do you see the same problem when running unmodified, stock firmware from the manufacturer?

Cheers from New York,
-Travis

iball, 2016-07-16:
It had the 13.09 firmware on it and it was fine out-of-the-box.
Don't see the problem at all with stock firmware.

iball, 2016-07-16:
Yup, just reflashed D013.009 back on it and the screen is now normal.
But putting the latest flashed firmware on it makes the screen flip upside-down and backwards.

d235j, 2016-07-16:
Does the D013.020 firmware work properly?

iball, 2016-07-16:
I have not flashed the D013.020 on it, just kept it on the .009 for now.
To be honest, I have no idea which of the "new" firmwares to use what with all this "new" and "old" vocoder stuff.  
To be honest, I have no idea which radio I have - the MD-390 no-gps-new-vocoder or the MD-390 no-gps-old-vocoder.

iball, 2016-07-16:
Nevermind, I've figured out the difference with the vocoders.
Really, Tyt?  Really? You make it this psychotic to figure out?

iball, 2016-07-16:
Works just fine with the new vocoder firmware D013.020.

aeickho, 2016-07-16:
Ok, Iball 
     With this Info: you have an "new vocoder" radio. 

And with the info that D013.020 is working correctly on your radio .. I'll tray to locate differences near the display communication. There is no flip with D013.020 on both MD380.

I'm thinking that the electrical display interface on md390 is changed. .

Please confirm the facts:
- md390 no GPS "now vocoder"
- D002.032 [experimental] works ... display flip
- NO problem with D013.020

Tnx for your assistance😉
Alex

d235j, 2016-07-16:
@aeickho it's very possible a vocoder-patched D003.020 will work too — chances are newer versions of the firmware have support for hardware updates.

aeickho, 2016-07-16:
My aplu bypass on 2.032. And D013.020  disabled not the complete alpu interface ... only functions the while 1 loops.  It is possible that the radio read  hardware info's /config from the dongle.

aeickho, 2016-07-16:
I'll tray to compare the low level interface to the display.. good: this behaviour contains all display output.

iball, 2016-07-17:
MD-390, no GPS, new vocoder - CONFIRMED

D002.032 (experimental) works....display is flipped and upside-down - CONFIRMED

No problems at all with the D013.020 firmware - CONFIRMED

iball, 2016-07-17:
Rebuilt and reflashed the latest firmware using latest md380tools as of a few minutes ago.
Still having the same flipped display issue.  Everything else in the experimental firmware works just fine.
Reloaded D013.020 on it and the display is fine again.

aeickho, 2016-07-17:
I see an difference between 
"F_1075_display_init"
D002.032 @ 0x08031fc4
and
D003.020 @ 0x0803356e

on D003.020 0x08033586 is a test 
  and  this set the display command 0x36 
to 0x8 (old D002.032) or 0x48

aeickho, 2016-07-17:
http://www.ncsys.co.jp/webshop/GTV350MPZI04(ILI9481).pdf
page 79

・ Bit B6 – Column Address Order
‘0’ = Left to Right
‘1’ = Right to Left

but... why is this clanged ?  :confused:

aeickho, 2016-07-17:
also command 0xfd is changed
old 0x40 now 0x40 or 0x4f

but I can't find documentation about the display command: **0xfd**

@iball .. you have interest? 
to look at a modified D002.32 (exp) whether this is "flip-probelm" then dissolved?
then I write a small patch for you...   
   or... for us to learn more about the differences of the radios
you can build fw?

iball, 2016-07-17:
Well, right now I have a Raspberry Pi 3 w/7" display that's "dedicated" to just doing stuff with the md380 firmware.  
I can run the md380tools set on it just fine and it flashes the radio, so I guess I could test out a patch from a separate code branch or something.
Just let me know what I have to do on my end and I'll get it done.

aeickho, 2016-07-17:
ok, i'll start now :)

aeickho, 2016-07-17:
please checkout:
https://github.com/aeickho/md380tools.git
branch :    display_flip_test

make clean :)

iball, 2016-07-17:
OK, on it.  I'll let you know what I find out.

iball, 2016-07-17:
Well, I'll be - that fixed it!
Of course, I wanted to check to see how your patch affects the regular MD-380 so I reflashed it as well and now the regular MD-380 is showing the backwards/flipped display.
No biggie, I can always just re-clone the latest from Travis' repo and get the MD-380 sorted out right.

Now for the hard part I guess.  
Adding in the code to detect the display chip in the radio so it knows whether or not to apply the patch for it?

aeickho, 2016-07-17:
tnx,
    for testing.
there are 2 temporary solutions _before @travisgoodspeed has an new experimental firmware based of the lasted official images._ 

1. try to understand the functionality to detect the hardware .. on D003.020  .. Handling of 0x2001dcf8
2. make an new additional global config setting to init the display ... but the initialisation is very early.. hopefully after the handling of the  read the additional global config .

@travisgoodspeed  what you mean?

travisgoodspeed, 2016-07-17:
Let's try and detect it for proper handling at runtime in 2.032, while we wait on proper portability between versions. 

Does anyone know where to order the upside-down hardware in America or Germany?

iball, 2016-07-17:
Travis,

I got mine from associatedradio.com (located in Overland Park, KS) from a ham fest yesterday in Warrensburg, MO.  
Not listed on their website just yet as they just got them in.  
Give them a call at 1-800-497-1457 and tell them you want one of the MD-390's that don't have a GPS.

travisgoodspeed, 2016-07-17:
Thanks, Chuck. I'll give them a call on Monday, and maybe they'll be able to find an old-codec GPS unit too. 

-Travis

aeickho, 2016-07-17:
I think I've found it:   the hw configuration is configured by the radio  tuning parameters

gn

iball, 2016-07-17:
I'm not Chuck, I'm just a friend of Chuck's.  
Chuck is well-known round these parts.  
All the locals know him.
He got together with some others and they made the www.byrg.net website and repeaters.

Anyway, thanks to you and @aeickho for resolving this flipped-display issue with the MD-390 non-GPS and I hope you can get it rolled into the main soon.

aeickho, 2016-07-17:
tuning parameter +0x1d &0x1

aeickho, 2016-07-18:
[image]

;) product of thinking errors

aeickho, 2016-07-18:
[image]

aeickho, 2016-07-18:
@iball please check with your "md390, old vocoder, no gps"

and reopen this issue if problem not solved

iball, 2016-07-18:
@aeickho I will check once I get home from work, probably in about 6 hours from now.
And this is the MD-390 with the new vocoder, no GPS, remember?
Here's what I'll do - I'll pull down the latest code from github and reflash both my MD-390 and my MD-380 and let you know what I'm seeing.  
Heck, I'll even throw in testing it with the MD-380 VHF unit as well. :-)

aeickho, 2016-07-18:
the patch is on travis/master 
  yes please check him with your MD-390 ... on MD-380 the patch make no changes, i hope so ;)

iball, 2016-07-18:
Nope.  The MD-390 screen is flipped and backwards with this update.  The MD-380 is fine as always.

aeickho, 2016-07-18:
this is bad

aeickho, 2016-07-18:
D003.020
```
|           0x08033586      6748           ldr r0, [pc, 0x19c]         ; [0x8033724:4]=0x2001dcf8                           
|           0x08033588      407f           ldrb r0, [r0, 0x1d]                                                              
|           0x0803358a      c007           lsls r0, r0, 0x1f                                                                
|       ,=< 0x0803358c      03d5           bpl 0x8033596               ;[4]                                                 
|       |   0x0803358e      0820           movs r0, 8                  ; 8                                                  
|       |   0x08033590      fff7d0ff       bl F_835_Write_Data_2display ;[3]                                                
|      ,==< 0x08033594      02e0           b 0x803359c                 ;[5]                                                 
|      |`-> 0x08033596      4820           movs r0, 0x48               ; 'H' ; 72                                           
|      |    0x08033598      fff7ccff       bl F_835_Write_Data_2display ;[3]                                                
|      `--> 0x0803359c      fe20           movs r0, 0xfe               ; 254                                                
|           0x0803359e      fff7c5ff       bl F_834_Write_Command_2display ;[2]     

```

New function for D002.032
```
/ (fcn) fcn.080a0348 20                                                                                                                                                                
|  ||||||   0x080a0348      044b           ldr r3, [pc, 0x10]          ; 0x80a035c:4]=0x2001d39b                                                                                      
|  ||||||   0x080a034a      5b7f           ldrb r3, [r3, 0x1d]                                                                                                                         
|  ||||||   0x080a034c      db07           lsls r3, r3, 0x1f                                                                                                                           
| ,=======< 0x080a034e      02d4           bmi 0x80a0356               ;[1]                                                                                                            
| |||||||   0x080a0350      0820           movs r0, 8                  ; 8                                                                                                             
| |||`====< 0x080a0352      91f713be       b.w F_835_Write_Data_2display ;[2]                                                                                                          
| `-------> 0x080a0356      4820           movs r0, 0x48               ; 'H' ; 72                                                                                                      
\  || `===< 0x080a0358      91f710be       b.w F_835_Write_Data_2display ;[2]   

```

aeickho, 2016-07-18:
@iball please try the next experiment

iball, 2016-07-18:
Did you do a fresh commit?

aeickho, 2016-07-18:
yes ... https://github.com/travisgoodspeed/md380tools/commit/6c476b270fa8281c5ea926ef1c42a42996405e51

iball, 2016-07-18:
Alrighty then, I'll flush my current md380tools directory and clone it new and try it out.
Gimme a few minutes...

iball, 2016-07-18:
Nope.  Still upside-down and flipped backwards.

aeickho, 2016-07-18:
ok, tnx for your assistance, I'll learn tomorrow more about the tuning parameter ..

iball, 2016-07-18:
No problem.  I'm around if you need further testing.

aeickho, 2016-07-18:
0x08020a72 @ D002.032 =0x2240b580 wtf

aeickho, 2016-07-18:
.... trouble with radare2, wrong offset calc :(

aeickho, 2016-07-18:
https://github.com/radare/radare2/issues/5347

aeickho, 2016-07-19:
`on D003.020 @ 0x8033586 .. 0x8033724 -> 2001dcf8`

iball, 2016-07-19:
@aeickho I can test this later today when I get off work in about 7 hours if you'd like.

aeickho, 2016-07-19:
Sounds great. Give it a go.

iball, 2016-07-19:
Testing it now.

iball, 2016-07-19:
Nope.  Still flipped upside-down and backwards.

aeickho, 2016-07-19:
... I'm waiting for radare2 update.
   tnx,alex

aeickho, 2016-07-19:
might find the Reading of spi flash after display init

aeickho, 2016-07-19:
@iball you are online and ready for more tests now?

iball, 2016-07-19:
Yes, I'm online and can do some more testing for you.

aeickho, 2016-07-19:
yes ... my idea is correct ...  display init for spi read

il'll fix it now

aeickho, 2016-07-19:
please check:
https://github.com/aeickho/md380tools.git
branch:  flip

aeickho, 2016-07-19:
no is _not_ working with md380

iball, 2016-07-19:
Aborting!  Cease-fire!  Red-two, this is Charlie-niner-two, clear and elevate all weapons and return to the baseline!  :-)
Let me know when you've got something to try for the MD-390.

[6 more replies]

commit 661f1794 (2016-07-18): fixed #178 flipped display on D002.32exp and old vocoder md390 no gps
closed 2016-07-20 by aeickho