travisgoodspeed/md380tools #180: Send message browse contact - radio reboot

iw9grl, 2016-07-17:
trying browsing contact to send a message the radio reboot

aeickho, 2016-07-17:
Hi Iw9grl,

iw9grl, 2016-07-17:
Hi alex
- Retevis Rt3 
- 3.008
- old vocoder
- uhf

thank you to you 
73 de IW9GRL

aeickho, 2016-07-17:
Am 17. Juli 2016 16:42:48 MESZ, schrieb iw9grl <notifications@github.com>:
>Hi alex
>- Retevis Rt3 
>- 3.008
>- old vocoder
>- uhf

... and experimental firmware from?

Very interesting... it is possible for you? to make an short video about this behaviour ?
I'll tray to repeat this on my radio.

Alex, df8av

iw9grl, 2016-07-17:
https://www.dropbox.com/s/0ndq1wmaafx23p9/VID-20160717-WA0021.mp4?dl=0

aeickho, 2016-07-17:
Crazy extreme crazy .

can you build yourself the experimental firmware from the github source? 

occurs the reboot also if you enters a message by the keyboard?

For my understanding... Please install the original D02.032 and check this behaviour.

Then reinstall D3.008 and check with this version .

Tnx, alex

iw9grl, 2016-07-17:
Hi Alex

I found the problem, too many contacts inside the 999 native positions .
I delete about 600 contacts and now works ..... very very strange !!!!!!
Probably the micro is not able to browse 999 contacts !!!!

thank you for your support
73

aeickho, 2016-07-17:
It the following test possible for you 

Put the big code plug into the radio 
And install the D03.020 into the radio 

Then we can learn:  if this bug fixed on newer fw versions 

Tnx

iw9grl, 2016-07-17:
where Can I found the 3.020 firmware?

aeickho, 2016-07-17:
Am 17.07.2016 um 19:12 schrieb iw9grl:
> where Can I found the 3.020 firmware?

D003.020.bin
https://archive.org/download/TYTMD380FW2/TYT-TFT-MD380-D3.20.bin
TYT-TFT-MD380-D3.20.bin
996864
38787da956234f9ec811cae08568f7bdb5f21b4070681ae689a0ba82c845f973

from:
https://github.com/travisgoodspeed/md380tools/blob/master/firmware/firmware_files.txt

iw9grl, 2016-07-17:
DOESN'T WORK ...

the 3.020 original firmware works, but after loading the md380 patched the browsing make a reboot !!!!

iball, 2016-07-17:
On my MD-380 (old vocoder) with the latest patched firmware (as of a minute ago), I have 532 contacts and it does NOT reboot or otherwise show this behavior when sending a message.
I also have the users.csv installed too.

iball, 2016-07-17:
Yup, that's what I was doing - sending a message.  Also tried it on my newly-flashed MD-390 and it does not reboot either.  
As was mentioned earlier, it may have just been the sheer number of contacts you had in that codeplug?

travisgoodspeed, 2016-07-18:
Perhaps duplicate contacts, or contacts with certain flags?

@iw9grl Could you please share a copy of your codeplug, as well as the exact text message recipient that triggers the bug?

aeickho, 2016-07-27:
@iw9grl I repeat the wish from travis ..   

I'll close this issue :)

iw9grl, 2016-07-27:
sure ... sorry for the delay, I will share it as soon as possible, I'm not at home ...... 73

aeickho, 2016-09-01:
@iw9grl :wave:

sijskes, 2016-10-07:
no response. reopen if not fixed.

closed 2016-10-07 by sijskes