travisgoodspeed/md380tools #181: Bootstrapping S013.020

travisgoodspeed, 2016-07-17:
This issue is to track bootstrapping S013.020 firmware with at least the following features.  Wherever possible, they should be implemented as C function hooks with an absolute minimum of assembly patching.  After S013.020 is targeted in this manner, we'll try adapting other releases.

Close this issue when the firmware boots with the following features:

- [ ] USB communications.
- [ ] Startup demo.
- [ ] Promiscuous modes.
- [ ] Menu items.
- [ ] Caller ID.  (users.csv)

burnersk, 2016-08-01:
S013.020 is only for GPS devices (MD-390 with GPS, not MD-390 without GPS, not MD-380).

Does this sum up into non-GPS devices (e.g. MD-380) are not supported anymore? Or does the new bootstrapping introduce device detection and (automatic) selection of either S013.020 or S003.020?

travisgoodspeed, 2016-08-01:
We will maintain support for old and new vocoder hardware as well as both GPS and non-GPS devices. 

S013.020 was chosen for this issue because it is the most different from 2.032, and by targeting it we will hopefully be able to link against all firmware revisions and targeting all hardware revisions.

aeickho, 2016-08-22:
the header of the "S" Type radios is changed ... 

```
 ../md380-fw --unwrap bin/S013.020.bin x
DEBUG: reading "bin/S013.020.bin"
WARNING: Funky header:
4f 75 74 53 65 63 75 72 69 74 79 42 69 6e 00 00

```
the "xor" is the same as the "D" Version

temporary --unwrap and then --wrap makes different version of the encrypted file.

aeickho, 2016-08-22:
I think ... this is not a problem ... @cr  can you confirm that?

aeickho, 2016-08-23:
I have an report ... from a md390, old vocoder, with gps user 
  ... my patched S013.020 (only alpu check disabeld) works with unmodified  ../md380-fw 
I'll push this patch this nearly.

aeickho, 2016-08-23:
I start to generate symbols_s13.020 with d13.020 as gold -> lot of matchs

aeickho, 2016-08-24:
- [x] bypass vocoder copy protection: added with  755d6e742b46bb6ac71a62d7a391e16dd94b313c

HB9TQJ, 2016-08-28:
I noticed one bug: If I assign 1750 Hz Tone to one button the secondary function (long or short press) does not work.

aeickho, 2016-09-03:
I think that I'm finish with the port on s13.020 in  some days.

aeickho, 2016-09-03:
@HB9TQJ ... Please open a new issue. ... Than I I'll label it with "bug on original s13.020 fw "

aeickho, 2016-09-06:
---- first test version now ... 4fdd6756f762a090f063f3eddd2ebb7a47f98dc4
+ usb hooks
+ nicer font
+ welcome message  (set intro screen to Char String)

flash please with `make flash_s13.020`

@ all:  please report ... i have no gps type radio to test

aeickho, 2016-09-07:
UPDATE: S13.020 port is finished now. but not tested yet. 

It works hopefully on the flowing radios: 

- md380-gps-oldvocoder 
- md390-gps-oldvocoder
- md380-gps-newvocoder 
- md390-gps-newvocoder

PD3LTX, 2016-09-08:
make flash_S13.020 works for my md390 with old vocoder with GPS (original firmware s003.012)
GPS world icon visible on display. GPS menu item visible. GPS function not tested. Still with Welcome logo.

Astralix, 2016-09-08:
Just a short rough test, but s13.020 seems to run fine on my MD-380G.
Thank you for the piece of fine work!

bendikz, 2016-09-08:
Tested here, seems to work as it should! Great work!
Only thing that I could think of would be the welcome logo. The M. Rogerbeep is slightly higher pitched, I can see if I can do a demo video here.

bendikz, 2016-09-08:
https://youtu.be/yVnGagVu3W8
Video demo of the MD390 firmwares S013.020 and D002.032 M.RogerBeep. 

1st is D002.032 ＞ S013.020
2nd is S013.020 ＞ D002.032

The S013.020 firmware sounds slightly higher pitched, also on D002.032 the M.RogerBeep setting also affects other sounds which, in my opinion, sounds better.

aeickho, 2016-09-14:
Can we say: the "S" Version is supported yet?

Doku moved to #292

bendikz, 2016-09-14:
I would definitely say that the S013.020 build is stable enough to pass, I have not bumped into any problems.

I got two MD390 GPS both running this firmware without problems. The GPS seems to work as it should too, tested both GPS Info and ViewGPS (from other radio)!

travisgoodspeed, 2016-09-14:
@bendikz Since you have a GPS radio handy, could you please check the features listed at the top of this issue?  I think that only the startup demo is still broken, but I'd like to know for sure.

bendikz, 2016-09-14:
I have conducted some more tests and you are right @travisgoodspeed. I have attempted to look into the boot logo as well however they have done something different with the newer reasons..

It seems like that USB Comms does not work 100% yet. I attempted to interpret the call log with md380-tool but nothing showed up (the radio interfaced with PC, showed up "Hooking calls!". Other features such as flashing firmware and csv and dmesg worked.  This test was done on a 64 bit computer running Ubuntu 16.04 (latest LTS).

- [ ]  USB communications. (partial)
- [ ]  Startup demo.
- [x]  Promiscuous modes.
- [x]  Menu items.
- [x]  Caller ID. (users.csv)

travisgoodspeed, 2016-09-14:
That's probably a bug in the call logging, rather than in the USB comms.  Perhaps @aeickho knows?

The startup demo is also broken in D13.020.  That issue is being tracked at #221.

bendikz, 2016-09-14:
That is likely it. I got a timeout error when I attempted to hook the call log, see http://pastebin.com/pnLET9Xb. This appeared after CTRL-C interrupting.

aeickho, 2016-09-22:
my md 380 g (with gps) has arrived. so i can test it.

@sijskes there are important things that I should transfer from "merge_d13.020" to merge_s13.020? 

I do not think so...

sijskes, 2016-09-22:
@aeickho if you can:
- [ ] hook all calls gfx_blockfill en gfx_drawbmp (so we have a central point to do draw intercepts)
- [ ] change eye symbol blockfill coordinate

aeickho, 2016-09-24:
Dmesgtail stops the menu task. 

Menu, channel switch and power off is not working.

I have a ram dump to the spi flash with inactive USB hooks and Our    USB transfer ram ... I think is it unused ..  

I dump ram to flash 5 sec after power on

Tomorrow .. I'll create an now issue for this problem

sijskes, 2016-09-24:
@aeickho i've merged two labels in flash.r for the S13.020 fw.

ispkorte, 2016-09-25:
if i flash RT8 s13.020 i get a lot of fails warning

snip /

arm-none-eabi-ar: creating libstm32f4.a
In file included from src/main.c:19:0:
src/gfx.h:81:2: warning: #warning TODO: define gfx_info [-Wcpp]
  #warning TODO: define gfx_info
   ^
In file included from src/usb.c:28:0:
src/gfx.h:81:2: warning: #warning TODO: define gfx_info [-Wcpp]
  #warning TODO: define gfx_info
   ^
src/usb.c: In function 'get_md380_dnld_tohook_addr':
src/usb.c:267:10: warning: 'ret' may be used uninitialized in this 
function [-Wmaybe-uninitialized]
    void * ret;
           ^
In file included from src/gfx.c:11:0:
src/gfx.h:81:2: warning: #warning TODO: define gfx_info [-Wcpp]
  #warning TODO: define gfx_info
   ^
src/gfx.c: In function 'gfx_drawtext4_hook':
src/gfx.c:268:2: warning: #warning should find symbol gfx_drawtext4 [-Wcpp]
  #warning should find symbol gfx_drawtext4
   ^
src/menu.c: In function 'create_menu_entry_rev':
src/menu.c:222:2: warning: #warning TODO find language menu on this 
firmware version [-Wcpp]
  #warning TODO find language menu on this firmware version
   ^
In file included from src/rtc_timer.c:14:0:
src/gfx.h:81:2: warning: #warning TODO: define gfx_info [-Wcpp]
  #warning TODO: define gfx_info
   ^
src/os.c: In function 'OSMboxPend_hook':
src/os.c:120:2: warning: #warning please consider finding mbox pointers 
for this firmware version [-Wcpp]
  #warning please consider finding mbox pointers for this firmware version
   ^
In file included from src/aes.c:11:0:
src/gfx.h:81:2: warning: #warning TODO: define gfx_info [-Wcpp]
  #warning TODO: define gfx_info
   ^
In file included from src/ambe.c:18:0:
src/gfx.h:81:2: warning: #warning TODO: define gfx_info [-Wcpp]
  #warning TODO: define gfx_info
   ^
In file included from src/dmr.c:25:0:
src/gfx.h:81:2: warning: #warning TODO: define gfx_info [-Wcpp]
  #warning TODO: define gfx_info
   ^
In file included from src/display.c:14:0:
src/gfx.h:81:2: warning: #warning TODO: define gfx_info [-Wcpp]
  #warning TODO: define gfx_info
   ^
In file included from src/console.c:9:0:
src/gfx.h:81:2: warning: #warning TODO: define gfx_info [-Wcpp]
  #warning TODO: define gfx_info
   ^
src/console.c: In function 'con_draw1':
src/console.c:127:9: error: 'gfx_info' undeclared (first use in this 
function)
          gfx_info.xpos = 0 ;
          ^
src/console.c:127:9: note: each undeclared identifier is reported only 
once for each function it appears in
src/console.c:132:17: warning: implicit declaration of function 
'gfx_drawchar' [-Wimplicit-function-declaration]
                  gfx_drawchar(' ');
                  ^
make[1]: *** [console.o] Fehler 1
make: *** [flash_s13.020] Fehler 2

/snip

-- 

mit freundlichen Gruessen / kind regards

Joerg Korte

Internet Business Consultant

	Vor dem Drucken dieser E-Mail denken Sie bitte
	 an den Schutz der Natur und unserer Umwelt!

-----------------------------------------------------------------
-- Linux wird nie das meist installierte Betriebssystem sein,   --
-- wenn man bedenkt, wie oft man Windows neu installieren muß! --
-----------------------------------------------------------------
--                       I S P - K O R T E                     --
-- Resellerservice of Web Server & Web Hosting & VoIP Systems  --
-----------------------------------------------------------------
Support Adresse Deutschland:

         ISP-Korte
         -Joerg Korte-
         Idafehn Nord 115
         26842 Ostrhauderfehn
-----------------------------------------------------------------
Service : +49 9005 102 895 0 (1.99 Eur/min)°
Telefon : +49 180-3118877-00 (9cpm)°
Fax : +49 180-3118877-10 (9cpm)°

Italy	+39 02 00 61 21 34
Germany	+49 49 61 98 54 58

° aus dem netz der DTAG / mobil auch mehr
-----------------------------------------------------------------
UId:  DE248861249
EORI: DE 406 2631 3424 3657
B-Netz-A: 	RegNr 	08/297
	ID 	108/177
-----------------------------------------------------------------
Homepage:
         http://www.isp-korte.de
                 unsere Resellerplatform
                 Webhosting preiswert
Messenger:
	ICQ     : 317316709
-----------------------------------------------------------------

sijskes, 2016-09-25:
@travisgoodspeed can we include the s13.020 build in the ci-builds?

sijskes, 2016-09-25:
@ispkorte the error you found was valid, but a separate issue from this issue where you have reported it, could you file a separatie issue instead next time?

ispkorte, 2016-09-25:
my clickfault at respond

sorry, no problem

Am 25.09.2016 um 09:31 schrieb Simon IJskes:
> the error you found was valid, but a separate issue from this issue 
> where you have reported it, could you file a separatie issue instead 
> next time?

-- 

mit freundlichen Gruessen / kind regards

Joerg Korte

Internet Business Consultant

	Vor dem Drucken dieser E-Mail denken Sie bitte
	 an den Schutz der Natur und unserer Umwelt!

-----------------------------------------------------------------
-- Linux wird nie das meist installierte Betriebssystem sein,   --
-- wenn man bedenkt, wie oft man Windows neu installieren muß! --
-----------------------------------------------------------------
--                       I S P - K O R T E                     --
-- Resellerservice of Web Server & Web Hosting & VoIP Systems  --
-----------------------------------------------------------------
Support Adresse Deutschland:

         ISP-Korte
         -Joerg Korte-
         Idafehn Nord 115
         26842 Ostrhauderfehn
-----------------------------------------------------------------
Service : +49 9005 102 895 0 (1.99 Eur/min)°
Telefon : +49 180-3118877-00 (9cpm)°
Fax : +49 180-3118877-10 (9cpm)°

Italy	+39 02 00 61 21 34
Germany	+49 49 61 98 54 58

° aus dem netz der DTAG / mobil auch mehr
-----------------------------------------------------------------
UId:  DE248861249
EORI: DE 406 2631 3424 3657
B-Netz-A: 	RegNr 	08/297
	ID 	108/177
-----------------------------------------------------------------
Homepage:
         http://www.isp-korte.de
                 unsere Resellerplatform
                 Webhosting preiswert
Messenger:
	ICQ     : 317316709
-----------------------------------------------------------------

sijskes, 2016-09-25:
@aeickho dmesgtail slows down D13 as well.

aeickho, 2016-09-25:
I can do that ... Ci-builds also with s13.020

ispkorte, 2016-09-25:
confirmed S13 RT8GPS

aeickho, 2016-09-25:
@sijskes  s13.020 is now include into the ci-builds #388

aeickho, 2016-12-09:
I'll close it now... gps radios are working

closed 2016-12-09 by aeickho