travisgoodspeed/md380tools #183: firmware is 404

dysinger, 2016-07-17:
```
<!DOCTYPE html>
<html lang="en">
<!-- __ _ _ _ __| |_ (_)__ _____
    / _` | '_/ _| ' \| |\ V / -_)
    \__,_|_| \__|_||_|_| \_/\___| -->
  <head>
    <title>Internet Archive: Page Not Found</title>
```

from 

```
$ make
cd firmware && make all
make[1]: Entering directory '/home/user/src/github.com/travisgoodspeed/md380tools/firmware'
curl -f -L $(cat firmware_files.txt | grep D002.022.zip | cut -f2) -o dl/D002.022.zip
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
make[1]: *** [Makefile:57: dl/D002.022.zip] Error 22
make[1]: Leaving directory '/home/user/src/github.com/travisgoodspeed/md380tools/firmware'
make: *** [Makefile:18: firmwares] Error 2
```

travisgoodspeed, 2016-07-17:
I'm unable to reproduce this bug.  Perhaps the Archive has a CDN that's not yet been updated?

Can anyone else reproduce this issue?

aeickho, 2016-07-17:
@dysinger 
please sync git repro

D002.022 is removed with dad421098ea22dbf2f72d1a282093cf7b7144dd1

travisgoodspeed, 2016-07-17:
Since the version has been removed from the Makefile, and it doesn't exist on the server to update, and also because we have no intention of supporting even older releases, I'm closing this issue.

closed 2016-07-17 by travisgoodspeed