travisgoodspeed/md380tools #184: Add option to disable on screen volume indication

M0IVQ, 2016-07-18:
Due to the way the volume setting is measured, the display will occasionally keep showing the current volume level without the knob being touched (usually if on the edge of the threshold between adjacent levels).

Close this issue when possible to disable on-screen indication of volume setting.

janipewter, 2016-07-21:
Yes please either fix the onscreen volume display or add an option to disable it completely. Can't believe TYT still haven't fixed this in firmware, it has been a bug since day one...

nivex, 2016-07-22:
Alternatively, could there be a constantly updating volume indicator in the top bar? Or has it gotten too full up there?

aeickho, 2016-07-23:
...  0x0800df84 @ d002.032 Volume_Menu

eb4fbz, 2016-09-04:
See https://github.com/travisgoodspeed/md380tools/issues/98 for hardware details on volume readout. Pin 23: ADC123_IN0, DC voltage 0.1 to 1.6V plus noise from DMR BB IC line output.

travisgoodspeed, 2016-10-01:
Merged into #370. Reopen only if there is active progress in adding such a feature.

closed 2016-10-01 by travisgoodspeed