travisgoodspeed/md380tools #185: powering down, final credits

aeickho, 2016-07-18:
PoC
https://flauschfunk.de/VID_20160718_221245.mp4
https://github.com/aeickho/md380tools/commit/a0546ca6ddb2035454575287ecea1805cf5632e7

lol
I can't added more contributors, temp. 
I have now idea to disable the radio showdown.

aeickho, 2016-07-18:
on 0x08043f72 @ D002.032 is managed the power off 
this code is not to be done yet... 
why the radio switched off??? :confused:

phl0, 2016-07-18:
Coole Sache :+1:

aeickho, 2016-07-18:
+ dog feeding

@dl8bb many thanks for the great assistance, I was able to fix the problem as soon as possible. Thank you!

iball, 2016-07-19:
Umm..that's a pretty long power-off sequence.  Please tell me you're shortening it down or making it optional in a menu item somewhere?

aeickho, 2016-07-19:
https://github.com/aeickho/md380tools/blob/final_credits/applet/src/gfx.c

Near. End of file

You can see the delay functions

Play with the function ;)

...
Btw ... Yes  that's not a good idea. It was training for me to learn more about the power off subsystem. 

But I think this is a good start point for your  experiments ... Have fun

iball, 2016-07-19:
I see all the "OSTimeDly(1000)" entries.  Hmmm...couldn't that be easier to change if they were all...aw, crap, my stupid brain forgets, a defined variable or something so we only have to edit it in a single spot at the top of the code?  Or am I missing something?

aeickho, 2016-07-20:
OSTimeDly(1000) is correct...  try it - play with him.

iball, 2016-07-20:
Line 191 needs a space between the words.

iball, 2016-07-20:
Going to play with this a bit.

First one to get it to run Doom wins. :-)