travisgoodspeed/md380tools #189: Display TG or Repeater name instead of number

iball, 2016-07-20:
I noticed in the db Makefile that there's a way to pull down the DMR-MARC repeater database.
If something similar could be done for Brandmeister repeaters, then why not see about uploading that into the radio as well as users.csv so it will display the TG or Repeater name so instead of "12345678 -> 3100" it instead says "12345678 -> USA TG" or something like that?

I know there's a limited amount of space but right now the users.csv is 2.7MB and the DMR-MARC repeaters.csv is 572K.
Of course, over time the users.csv will grow and once it gets over 3.5MB then this won't work so..yeah.
Wish list it as an option?

aeickho, 2016-07-20:
Is this Issue identical to #136? I'm not an DMR Radio User ....

iball, 2016-07-20:
Actually, you're right.  
Excuse me while I get my time machine and go back and take basic reading comprehension in grade school again, 'cause I'm a big dummy.

travisgoodspeed, 2016-07-20:
Closing as a duplicate of #136.

closed 2016-07-20 by travisgoodspeed