travisgoodspeed/md380tools #191: make flash fails fatal error

davetelling, 2016-07-21:
Sigh... I'm trying to make the firmware flash in Mint Linux. If I run:
sudo make flash or
sudo make clean flash
I get this error both times:
/usr/lib/gcc/arm-none-eabi/4.9.3/include/stdint.h:9:26: fatal error: stdint.h: No such file or directory
 # include_next <stdint.h>
I look in /usr/lib/gcc/arm-none-eabi/4.9.3/include/ and there is, in fact, a file called stdint.h
So, why the fatal error?

iball, 2016-07-21:
Did you install all of the following?  

apt-get install gcc-arm-none-eabi binutils-arm-none-eabi libnewlib-arm-none-eabi \
                libusb-1.0 python-usb

davetelling, 2016-07-21:
yes, I installed as the webpage showed. I mainly don't understand why the script says that the file doesn't exist when it actually does.

davetelling, 2016-07-21:
I'm an idiot somehow, my blurry eyes totally missed libnewlib-arm-none-eabi package!
installing that fixed the problem. Consider my forehead slapped! Doh!

closed 2016-07-21 by davetelling