travisgoodspeed/md380tools #192: MD380 Experimental Firmware‏

M6NBP, 2016-07-22:
Hi

I was wondering if you could pass this onto whoever writes the Experimental Firmware.

Under Contacts could we have a new section for Talk Groups so we could call up any TG with out having to add them in as channels.

I have done it via the contact list, but the window stays up so blocking the main screen and when it does go away your no longer on that TG you wished to use.

If a new transparent window could be made that stayed open until you closed it, this would be a big step forward and make using any Talk Group usable with out having to program so many channels into the radio.

Also many thanks for the Experimental Firmware and all your hard work.

Many thanks

Norman

travisgoodspeed, 2016-07-22:
This would be handy, especially if we eventually add a GPS-driven tower database similar to our User.csv database.

@aeickho In your menu reverse engineering, have you figured out how the Contacts menu works?  That is, which function handler is triggered when the PTT button is pressed as a contact is selected?

aeickho, 2016-07-23:
re: not directly - but I'm thinking to know the location.  

I will look at me when we have done #181 and d013.020

F4EGX, 2016-08-27:
Hello,

It's possible to add a menu to change TG 
in Utilities/Program_Radio add something like "8 change contact"?
73
F4EGX

travisgoodspeed, 2016-10-01:
Merged into #370. Reopen only if there is active progress in adding such a feature.

closed 2016-10-01 by travisgoodspeed