travisgoodspeed/md380tools #193: md380-tools ... new command: restart radio in bootloader mode

aeickho, 2016-07-22:
I wish me:
 a new "md380-tools" command to restart the radio in the bootloader mode. 

so "make flash" is possible without pressing a button. 

-alex

dl8bb, 2016-07-22:
Would be very useful.

birko
--

travisgoodspeed, 2016-07-22:
There's a nice generic way to do it:

1) Disable interrupts.
2) Erase the block at 0x0800C000, so that the interrupt table is empty and the bootloader won't start the main application.
3) Branch to the address stored in 0x08000004, which is the bootloader's RESET vector.
4) The bootloader will then launch and wait for comms, rather than branch into the main application.

We could also do a late call into the bootloader, which would not require erasing part of the main application, but would require compatibility with specific bootloader versions.  I don't know whether there are multiple bootloaders or just the one that I used for the initial jailbreak.

Cheers from New York,
--Travis

aeickho, 2016-07-22:
tnx Travis, 
I'll try it tomorrow.
gn
but .... after flashing power off/on in necessary... ???

No....  after flashing fw `./stm32-dfu go`  starting the main application on 0x08000000 
 ---> ok not really an jump ... more a crash with reboot

travisgoodspeed, 2016-07-30:
`stm32-dfu` is just for talking to the Mask ROM bootloader, which is different from the MD380's Flash ROM bootloader.  The Flash bootloader doesn't understand the same commands, and won't allow branching to arbitrary addresses.

aeickho, 2016-07-30:
yes I know ... it was only a fast try to reboot the stm32 after flashing ... my bee  wrong command :)

aeickho, 2016-07-30:
@travisgoodspeed you have sources to md 380 similar bootloader. to find a "go" similar function

travisgoodspeed, 2016-07-30:
Self assigning.

I'll close this issue when a command exists to either enter the Flash ROM bootloader (without holding buttons) or the Mask ROM bootloader (without inteference from RDP).

mach327, 2018-06-19:
So it still needs to be nicely integrated with `-dfu`'s `upgrade`, but you can now (after flashing again with these patches):
```
md380-tool reboot_to_bootloader
md380-dfu upgrade some.bin
./stm32-dfu go 0x0800C000
```
which successfully reboots to bootloader, flashes, and reboots into the application (even if it's not correct, it Works On My M;achine (tm).

closed 2018-06-19 by mach327

travisgoodspeed, 2018-06-19:
Excellent!

mach327, 2018-06-20:
pull request opened for auto-upgrade feature: #880 

Please help test and make sure it's somewhat sane. If there's no problems reported, I'll merge this week.

mach327, 2018-06-20:
I also have a couple ways verified to be able to transmit from an attached MD380 VHF without getting too much RFI into the USB port:
* A 30db attenuator before the antenna is verified to work great.
* Two large ferrite chokes and the usb cable wrapped through the chokes four times is also satisfactory on high power.
* A single large ferrite choke with the cable wrapped through four times is sufficient for my setup, but only when transmitting at low power.

commit e1bce347 (2018-06-19): reboot_to_bootloader dfu command implemented, fixes #193