travisgoodspeed/md380tools #194: UserCSV Window

M6NBP, 2016-07-22:
UserCSV Window pop up 

Could this be made full screen with larger text?
Also can the time it stays up be made longer?

DK3DO, 2016-07-23:
I'd like also to attach to "this **wishlist**";

**-would be nice to also shown the current channel name in the green csv-window,** 
because if device scanns and stops for a conversation, the green window overlaps the channel names and so nobody can identify the channel, the conversation came from.

Btw.
**Which locale settings I needed to show the "german umlauts" (äöüß) correctly?**
(I modified the users.csv byself but couldn't generate the umlauts correctly at md380)

aeickho, 2016-07-23:
- there are no "German Umlauts" in the MD380 Charset  .. see also  #73 - 
add "additional chars" currently  an hard job 

-  current channel name in csv window ... currently i have no address with the channel name but the channel nr...I'll commit it in the near future.

- bigger font, there are a lot of requests from many user ....  it's is on my todo list on top :) 
but there are so many jobs on my list  on top :)

DK3DO, 2016-07-23:
_there are no "German Umlauts" in the MD380 Charset_
No Problem, I could use "ue" for "ü" for instance.

_hannel nr...I'll commit it in the near future._
That would be also very nice.

_bigger font_
I think, bigger fonts could become areal challenge. I need all 4 lines, because my csv shows call sign, state, full name and location. I can well understand the arguments for the bigger font fraction, too.
So, please - if you change the setting to bigger fonts, please include a switch (in patchfile or somewhere else), to stay by the actuelly size.

travisgoodspeed, 2016-09-14:
Close this issue when #281 is successfully merged.

travisgoodspeed, 2016-10-01:
Closing this issue.  The new RX screen looks pretty damned spiffy!

closed 2016-10-01 by travisgoodspeed