travisgoodspeed/md380tools #195: dnld_tohook? I can't find it

aeickho, 2016-07-22:
I'm going crazy .... Where is "dnld_tohook" initial describes ... for hours I'm looking for it.

I can nowhere be found 0x808ccbf or 0x20000e9c in this fw ...

I doubt my ego :pensive: 

But I've learned a lot about today STM32 usb!    :grinning:

aeickho, 2016-07-23:
insomnia..,
    .... In the md380 Bootloader@0x8000000 .... I'll look after  sleeping

aeickho, 2016-07-23:
I have discovered a new track. USB is hard, but nice, good training

travisgoodspeed, 2016-07-28:
As the authors forked a reference design, most of the source code for the USB/DFU stack is available from STMicro in the form of example code for the STM32F407.  Reading that might help.

For patching in future revisions, I think we'll have an early code hook that scans all of SRAM for 32-bit pointers to the USB handler functions.  If we are careful, there should be no false positive matches.

closed 2016-07-28 by aeickho