travisgoodspeed/md380tools #200: Mysterious reboots

iball, 2016-07-27:
So if I leave my MD-380 or MD-390 alone on the table, after about 20-30 minutes it will mysteriously reboot itself.  
Running the firmware on it compiled from the latest source.
Possible memory leak?  
Any way I can capture some sort of log of this when it occurs?
Scooby Doo, where are you?

aeickho, 2016-07-27:
I can confirm this behaviour.  

There are some reports from different OMs, but also some reports - that the reboot occur with d002.32 Vanilla. 

For good reporting of  mysteriously reboot: With enabled power on password... the boot stopped and ask for password.

My test settinging (d002.32 + MD380tools-Master-Head) : 
on free dmr simplex channel , only power on... -> mysteriously reboot

aeickho, 2016-07-27:
@iball please install d002.32 Vanilla on your radio and look for reboots.

aeickho, 2016-07-27:
I thinking about my "idea"  taskmonitor   ... > possible new Issue ;)

- hook on beeptone process  (runs continuously, very small task)
- read last entry from spi log write 
- dumps stack position (used stack) from all task to spi flash + uptime cnt

aeickho, 2016-07-27:
I am surprised that so few messages come to re-boots.

phl0, 2016-07-27:
I currently only have the latest mod FW at hand. Is that in scope of a
reboot test?

aeickho, 2016-07-27:
@phl0 you have  no notice from the reboots?

phl0, 2016-07-27:
I am not sure if I had the radio run for such a long time :)
Only used it for testing these days as my MMDVM mit GM1200 is not really
working due to the problems with GM1200 as RX. :)

Will let the radio run and keep an eye on it tonight.

aeickho, 2016-07-27:
@phl0 with enabled beep tones the is also a good solution to hear an reboot

iball, 2016-07-27:
@aeickho I'll reflash the MD-380 with the original 2.032 and keep the MD-390 on md380tools.

phl0, 2016-07-27:
> @phl0 <https://github.com/phl0> with enabled beep tones the is also a
> good solution to hear an reboot

Alright. Tnx fer tip. Will do

iball, 2016-07-27:
MD-380 was reflashed to 2.032 2.5 hours ago and powered on.  No reboots so far.
MD-390 was powered on at the same time as well.  No reboots so far.
But I know they're coming...

iball, 2016-07-27:
Had both radios powered on for the last 8.5 hours with no mysterious reboots.

travisgoodspeed, 2016-07-27:
Our patches do not use a heap, but it's possible that the firmware uses a heap which we haven't identified yet. Let's check to see if a painted region is growing?

phl0, 2016-07-30:
I cannot reproduce this issue on latest patched firmware.
Had it running for several hours on the last 2 days now. Not a single
reboot here ....

iball, 2016-08-01:
I saw it again last night on my MD-390.  Left it on the entire day and it it was fine for 8+ hours then it just rebooted itself.

paulomartinspt, 2016-08-03:
Mine just did it last night for the first time :(
Latest firmware available, installed on 01/08.

aeickho, 2016-09-01:
@iball @phl0 @paulomartinspt 

can someone tell if the reboot occurs with the d13,020 still on ....

iball, 2016-09-05:
@aeickho I have not seen it reboot since reflashing either my MD-380 or MD-390(no GPS) with the new d13.020 firmware.

travisgoodspeed, 2016-09-13:
Closing this issue on the assumption that it only affects 2.032.  Please reopen it if you witness the reboots on 13.020.

closed 2016-09-13 by travisgoodspeed

eb4fbz, 2016-09-23:
It happened to my MD-390 GPS with stock S13.020 last saturday after +6h running. It suddenly rebooted.

Also got it locked in a busy channel as usual, but turning the channel selector had no effect, i had to power it down. All this with stock S13.020, so i decided to reopen this issue.

nokiaman2002, 2017-07-18:
I think I have found a possible case of the random reboot. At my home QTH it reboots randomly back to enter password. However it only does this at my home QTH and ONLY on a certain scan list. I think the reboot occurs when the radio hears an ALL CALL and PROMISCUOUS MODE is turned on. I will do some more investigation.

travisgoodspeed, 2017-07-18:
@nokiaman2002 Can you confirm this by transmitting an ALL CALL in promiscuous mode?

nokiaman2002, 2017-07-18:
I can transmit ALL CALL fine. I am doing some more testing as this only happens when receiving a local user that uses ALL CALL as the TX contact. Initially I thought it was to do with scanning but it is not.  More to follow

nokiaman2002, 2017-07-18:
https://youtu.be/tWQ8LpAxlsw this should explain more

travisgoodspeed, 2017-07-18:
Please use words instead of video.  Is the issue reproducible now?

nokiaman2002, 2017-07-18:
Yes. When RT3 receives an ALL CALL it reboots. Regardless of promiscuous mode. It does not demodulate the audio also. Testing on stock firmware shortly.

travisgoodspeed, 2017-07-18:
Really?  I use All Call all the time without trouble, and I think some countries use it instead of 99 for their simplex channels.

Please test both stock firmware and a stock codeplug to see if it's related to some specific configuration.

nokiaman2002, 2017-07-18:
Problem no longer persists using stock firmware D013.032. Strange.

nokiaman2002, 2017-07-18:
reflashed experimental firmware from another source, same issue. I wonder what it could be? Its not a major problem as such but it needs looking at I think. It is the Retevis RT3 I have.

nokiaman2002, 2017-07-18:
And just to be sure, I have changed the frequency, CC and TG as well. Still the same.

travisgoodspeed, 2017-07-18:
Exactly what should I do to reproduce the problem?  Step by step, with all version numbers specified.  Please be very specific, as in the twelve months since this issue was opened nothing useful has come of it.

nokiaman2002, 2017-07-18:
The problem is that most people do not report bugs, they just moan and whinge about them to their fellow amateurs instead of telling the developer. When I got the radio the details are: MODEL: MD380, CPS SOFTWARE VERSION: V01.30 HARDWARE VERSION V01.00, MCU VERSION D013.032. It ran fine on stock firmware. Now on MD380 tools using "Tytera Flash Tool" by KG5RKI it is the same apart from CPS SOFTWARE V01.32 , mcu = d013.020. I have tried numerous other versions of the firmware and still the same.

travisgoodspeed, 2017-07-18:
The KG5RKI flash tool uses DL4YHF's fork of md380tools.  Do you have the same problem in the main branch?

Binaries are available at http://md380.org/releases/daily/

nokiaman2002, 2017-07-18:
Yes I have tried that one, still the same. I am going to try an older firmware from May 2017 to see if the problem is still there.

nokiaman2002, 2017-07-18:
Odd, firmware from 13th May, still the same.

travisgoodspeed, 2017-07-18:
Could you please compose step by step instructions for exactly how to reproduce this problem, step by step and with all version numbers specified?

nokiaman2002, 2017-07-18:
Well you would need an RT3 HW v 1.00 to start with, on stock it should be D013.032 and I used MD380 CPS v 1.32. Create a simplex frequency with anything or nothing in the RX group list. Set TX contact to "ALL CALL" 16777215. Load any version of the MD380 tools and call the RT3 using ALL CALL from another radio. The RT3 should not demodulate the audio, the light on top will indicate frequency in use then after a few seconds the RT3 should reboot.

travisgoodspeed, 2017-07-18:
I am unable to crash md380tools on S13.020 with an All Call transmission on 441MHz simplex.  I don't have a non-GPS model handy, but will try to replicate it with D13.020 with one of those tomorrow.

nokiaman2002, 2017-07-19:
I am keen to know the results of that. Thanks for your time.

nokiaman2002, 2017-07-19:
I have an update! I flashed the radio using the Vmware image from KD4Z and the problem has gone away!

KD4Z, 2017-07-19:
My VM builds from 'master' of travisgoodspeed/md380tools, except for my customizations in display.c   My guess is there is an overflow in **draw_rx_screen()** or possibly **draw_statusline_hook()**  I have refactored the logic quite a bit in there and may have inadvertantly bypassed the condition.     I have two MD380 radios so I can set up using the ALL CALL method instead of 99, and try to get it to break.

nokiaman2002, 2017-07-20:
I note something else about this VMWare flashed firmware: The volume indicator is present on the screen when changing volume. This may be irrelevant or it may not be. When the "ALL CALL" problem was present this volume indicator had been removed in the firmware.

travisgoodspeed, 2017-07-20:
Today I flashed the latest patched version of D13.020 to a radio and transmitted an All Call message.  No crash.

Leaving this issue closed until there's a bug report that we can reproduce.

KD4Z, 2017-07-21:
I did the same.  Flashed D13.020 03a05ff to MD380.  Configured simplex channel on ALL CALL of 16777215 in this and another MD380.   Using different DMR Ids in each radio.   No reboots occured.

I did notice a bug in Tyt CPS 1.32 however.  When adding the All Call contact, the entry field for Call ID changes the number from 16777215 to **16777415** when you press Tab or Enter.  I did this twice thinking I fat-fingered it.  Nope.   Bug in CPS.  You have to go back and edit the 4 back to a 2.     You might make a note of this when attempting to reproduce this issue.  I was unable to reproduce the issue in current master and in my VM with the display enhancments.

f5vmr, 2019-02-04:
On 23rd January I use the KG5RKI tool to update MY RT3 (13.020 No GPS), and since then randomly a reboot occured. So this morning I did it the hard way using the latest version from MD380.org/tools. Within 10 minutes I got a mystery reboot. I didn't check the screen or my repeaters dashboard to see what activity might have caused it, if there is a belief that ALL CALLS might be the issue,