travisgoodspeed/md380tools #204: Not updating users.csv in Linux

iball, 2016-07-27:
When executing "sudo make flashdb" if there is already a users.csv file present in db it will use it instead of downloading a fresh copy pulled from the DMR-MARC database.

See screenshot for output.

[image]

iball, 2016-07-27:
As you can see, a sudo make clean does not delete the users.csv file in the db directory.

[image]

travisgoodspeed, 2016-07-27:
This is intentional, so that development can be performed offline without erasing the database.  Similarly, we don't delete the firmware images on a global `make clean`, as you might need them for the next build.  You can always grab the new database by running `make clean all` in `db/`.

Closing issue.

--KK4VCZ

iball, 2016-07-27:
OK, I completely understand this however perhaps something in the README about it should be noted?
Otherwise unsuspecting users not familiar with it - not all hams are real familiar with Linux and make tools - will never get users.csv updated properly and possibly open future tickets on "ID not found in users.csv".

travisgoodspeed, 2016-07-27:
Fair enough.  Reopening the issue.

Close this issue when the README.md mentions the database rebuild procedure.

iball, 2016-07-27:
@travisgoodspeed I wrote a small blurb about it.

aeickho, 2016-07-27:
@iball your patch #205 ... please remove the sudo ....  
Do you know https://github.com/travisgoodspeed/md380tools/blob/master/99-md380.rules
with this rule anybody on your system have access to flash the radio
this is also an note for the documentation

iball, 2016-07-27:
@aeickho Ah yes, excellent point.  
I'll work on those changes and include the blurb about ensuring that the 99-md380.rules is placed in /etc/udev/rules.d/ and remove the sudo.

iball, 2016-07-27:
OK, changes made.  Please review when you get time.

travisgoodspeed, 2016-07-28:
Build instructions have been updated with 7f4b052.  Closing issue.

closed 2016-07-28 by travisgoodspeed