travisgoodspeed/md380tools #208: PTT Locked after 10mins of inactivity

IZ2QDH, 2016-07-29:
I've noticed that if the 380 does not receive any signal for 10 minutes goes into some "power saving" mode that block PTT.
If you press ONLY PTT does not transmit, you have to twist the channel knob back n forth to "unlock" the radio and being able to transmit.
It happens only on digital mode, both with repeaters and simplex channels.
I've made a explanatory video: [https://drive.google.com/file/d/0B6jrNOKYX_1PS29YRUJOYUpnb3c/view?usp=sharing](url)

73

travisgoodspeed, 2016-07-30:
I'm unable to reproduce this either in stock 2.034 firmware or in our patched 2.032 firmware by waiting 11 minutes on an unused digital simplex channel.  Is there something strange about your codeplug?

Do you see the same problem when running the official 2.032 firmware?

--Travis

IZ2QDH, 2016-07-31:
You were right Travis, i have set "admin critera" to "channel free", now i have set it to "always" and the defect disappeared...but could be also a firmware problem because if the radio does not receive any signal the channel is free but it acts like it's busy.
Fixed both on simplex and repeaters

M0IVQ, 2016-07-31:
Maybe this should be added to issue #120 - Not only is admit criteria not respected, but also has this side effect.

aeickho, 2016-09-01:
is this issue with d13.020 actual ?

travisgoodspeed, 2016-10-01:
As it's been a month without confirmation that 13.020 is afflicted, I'm closing the issue.  Reopen this ticket if we find a way to reliably reproduce the problem.

closed 2016-10-01 by travisgoodspeed