travisgoodspeed/md380tools #209: RT3 starting firmware before apply pathed

iw9grl, 2016-07-29:
Hi to all

I don't understand if it is important starts from a particular genuine firmware before apply the pathed one. Could you explain it please ?

73

Astroscope, 2016-07-30:
I'm pretty sure you may flash any appropriate firmware regardless a currently installed as it won't affect the bootloader. So you may flash any factory firmware over any factory or experimental and you may flash an experimental one over any factory or experimental. In particular you won't patch your installed firmware with md380tools but will flash the radio with a new patched one.

travisgoodspeed, 2016-07-30:
Yep, our patched firmware is a complete replacement.  You install it just like the factory images, and you uninstall it by installing a factory image.

Closing issue.

closed 2016-07-30 by travisgoodspeed

iw9grl, 2016-08-01:
tnx
73