travisgoodspeed/md380tools #212: Cannot creat Fiemware on Windows

IK8JHL, 2016-08-04:
Im using cgwin but i cannot create new firmware at end i have an error

"arm-none-eabi-gcc -g -O2 -Wall -mlittle-endian -mthumb -mcpu=cortex-m4 -mthumb-interwork -mfloat-abi=hard -mfpu=fpv4-sp-d16 -ffreestanding -nostdlib -Iinc -Iinc/core -Iinc/peripherals -fshort-wchar -c -o misc.o src/peripherals/misc.c"
arm-none-eabi-gcc.exe: error: CreateProcess: No such file or directory
If i check in the folder" src"  really i  read no directory only file. Therfore i have tryed "find" in windows and windows find "peripherals" folder in md380tools/applett/src but i continue to dont read.......
I have all visible in " folder option" all visible 
Some have experience? , im sure to solve if using linux but for now i have only windows 
Thanks

M0IVQ, 2016-08-04:
You need to install the arm-none-eabi-gcc toolchain under windows. If you find the correct line in the windows instructions, you'll find a link to it. You'll need to restart your PC after from my experience.

IK8JHL, 2016-08-04:
Thanks Simon , but i have installed and add enviorment path"C:\Progra~\GNU Tools ARM Embedded\4.8 2014q1\bin" 
As you can see  dont say "command not found" but "error: CreateProcess: No such file or directory" i think because cannot found directory "peripherals"

M0IVQ, 2016-08-04:
Is there a typo above or is the path entered exactly as above? Progra~ should be Progra~1. However, mine works with Program Files in the path variable with Cygwin.

IK8JHL, 2016-08-04:
No its only a mistake to copy .I have tryed on more pc seven and 10 but all the some problem....i think some problem of compatibility because its italian language ...i dont have any idea. This is the lis in cgwin:
Francesco@Francesco-PC /cygdrive/c/Users/francesco/Documenti/git/travisgoodspeed/md380tools
$ make flash dist
cd applet && make flash
make[1]: ingresso nella directory "/cygdrive/c/Users/francesco/Documenti/git/travisgoodspeed/md380tools/applet"
make -C lib
make[2]: ingresso nella directory "/cygdrive/c/Users/francesco/Documenti/git/travisgoodspeed/md380tools/applet/lib"
arm-none-eabi-gcc -g -O2 -Wall -mlittle-endian -mthumb -mcpu=cortex-m4 -mthumb-interwork -mfloat-abi=hard -mfpu=fpv4-sp-d16 -ffreestanding -nostdlib -Iinc -Iinc/core -Iinc/peripherals -fshort-wchar -c -o misc.o src/peripherals/misc.c
arm-none-eabi-gcc.exe: error: CreateProcess: No such file or directory
make[2]: *** [Makefile:34: misc.o] Error 1
make[2]: uscita dalla directory "/cygdrive/c/Users/francesco/Documenti/git/travisgoodspeed/md380tools/applet/lib"
make[1]: *** [Makefile:50: lib] Error 2
make[1]: uscita dalla directory "/cygdrive/c/Users/francesco/Documenti/git/travisgoodspeed/md380tools/applet"
make: *** [Makefile:20: flash] Error 2

Francesco@Francesco-PC /cygdrive/c/Users/francesco/Documenti/git/travisgoodspeed/md380tools
$

M0IVQ, 2016-08-04:
You need to use either:

Make clean flash (flashes firmware to radio)
Or
Make clean dist (creates firmware file for tytera flashing software)

Make clean flash will require pyusb and libusb to work.

iball, 2016-08-04:
I just compiled it fine using the new Bash shell in Windows 10 Redstone.  Did not install cygwin or anything other than doing the usual "sudo apt-get install" stuff from the command line.  
The firmware compiled successfully on it.  
You can't flash the radio with it just yet, but you can completely compile the firmware and then use the official Tyt firmware updater to flash it.  Trying to flash it or the users.csv file results in a seg-fault error.
All using Windows built-in tools.  
Wow.  You now basically have a command-line Ubuntu Linux shell inside Windows 10 now.

So if you're having issues with Cygwin on your Windows 10 PC, install the latest Win10 upgrade and then install the new bash shell and apt-get install your way to at least compiling the firmware fairly easily.

travisgoodspeed, 2016-08-12:
Closing this issue as a support request.  Those should go to the mailing list, rather than the issue tracker.

(Because this is ham radio, feel free to keep the conversation going until your problem is solved.)

Cheers from New York,
--KK4VCZ

closed 2016-08-12 by travisgoodspeed