travisgoodspeed/md380tools #218: Bootstrapping D013.020

aeickho, 2016-08-22:
re pull request #217

build and flash with  `make flash_ng` the D013.020.

Some notes:
..................
 sapplet.getadr("dmr_apply_squelch_hook"),0);
 merger.hookbl(0x0804105a, sapplet.getadr("dmr_apply_squelch_hook"),0);

- [ ] check monitor mode

..........
#These aren't quite enough to skip the Color Code check
in patch.py

- [ ] add Color Code patch to d13.020_patch.py

........
beep tone setup has been changed
- [ ] adjust beep tones

........
movement from applet/src/dmesg.h
from 0x2001d500 to 0x2001f700

- [ ] hopefuly free  ... please check stability 

.....
applet/src/usb.c

- [ ] add nicer variable names

......  
../../md380-gfx --firmware=patched.img --gfx=0x80f9ca8-troopers-color.ppm  relocate
dount work with d13.020

- [ ] look for the problem (img location) OR add an new bmp with ppm2h.py

........
travis test, temponary all tests in all cases: green

- [ ] better tests

.....
applet/src/rtc_timer.c
MD380_FONT_SMALL + MD380_FONT_NORM

- [ ] move to symbols_

....
reformat symbols_d13.020

- [ ] synced with d02.032

...
@travisgoodspeed  tnx for your tool symgrate
...
30h working hours
..

iball, 2016-08-22:
@aeickho FYI: I've flashed this to both my MD-380 and MD-390-no-GPS.
Let me know if there's anything I can test out for you.

aeickho, 2016-08-24:
Some notes /todos:
..................
 sapplet.getadr("dmr_apply_squelch_hook"),0);
 merger.hookbl(0x0804105a, sapplet.getadr("dmr_apply_squelch_hook"),0);

- > check monitor mode

..........
#These aren't quite enough to skip the Color Code check
in patch.py

- add Color Code patch to d13.020_patch.py

........
beep tone setup has been changed
-  adjust beep tones

........
movement from applet/src/dmesg.h
DMESG_START from 0x2001d500 to 0x2001f700

-  hopefuly free  ... please check stability 

.....
applet/src/usb.c
-  add nicer variable names

......  
../../md380-gfx --firmware=patched.img --gfx=0x80f9ca8-troopers-color.ppm  relocate and write 
don't work with d13.020

-  look for the problem (img location) OR add an new bmp with ppm2h.py

........
travis test, temponary all tests in all cases: green

-  better tests

.....
applet/src/rtc_timer.c
MD380_FONT_SMALL + MD380_FONT_NORM

- move to symbols_

....
reformat symbols_d13.020

- synced with d02.032

...
micbargraph screen update not ok

.....
monitor mode "eye" screenupdate not ok

.....
@travisgoodspeed  tnx for your tool symgrate
...
30h working hours - ( pull requst #217)
..
This Issure can be closed, when D013.020 is default

travisgoodspeed, 2016-08-24:
The color code is enforced both by firmware and by the C5000 baseband chip, so I'm cutting that as a requirement for closing this issue.

aeickho, 2016-08-24:
@travisgoodspeed can you check the monitor mode patch? I have no test environment ready...

aeickho, 2016-08-24:
@travisgoodspeed "check new DMESG_START address" is only a wish - that somebody is confirming my view. Please look at this.

aeickho, 2016-08-25:
check new DMESG_START address ... radio is working fine ...   Ram address ok :)

burnersk, 2016-08-26:
I would like to test this firmware on my MD380 (old vocoder). How do I produce the "md380tools-XXXX-XX-XX" directory structure using the D013.020 release (since it is not the default)?

aeickho, 2016-08-26:
On the md380 main directory
make clean
make flash_ng

- Alex

Tnx for testing

burnersk, 2016-08-26:
> On the md380 main directory
> make clean
> make flash_ng

I need the folder structure with the Windows updater application and the firmware binary. Unfortunately I cannot attach the radio to my build machine directly. So I have to transfer the firmware (or the entire folder) to a Windows machine.

I am assuming that the firmware would have the following path (this file to upload):

```
applet/experiment.bin
```

Am I correct?

burnersk, 2016-08-26:
**Error report** MD-380 (guessing: old vocoder - how to determine?)

Source: 660ef4e6cd24f516cd6e3c43b57ca75cd4312ac5

Environment: After flash_ng and flashing the Ham-DMR UserDB to SPI Flash. Radio is turned on and in default mode.

After 5-6 seconds when selecting a channel the signal strength bars (not the antenna symbol) and the promiscuous symbol are gone. The actual effects (async so cannot connect to relay, promiscuous is really off now) are currently not investigated.
Alternative measure: after 3 text scrolling iterations.

DL9LBG, 2016-08-26:
Not only the beep tones are too deep in frequency (which doesn't matter much to me) but also the 1750 Hz tone for european FM repeaters (assigned to the side buttons in CPS 1.32) is much too low (I think about 800 Hz or so).

73s de Michael DL9LBG

aeickho, 2016-08-26:
Hi Sören,
      I'm unsure if I understand your mail correct.

Correct is: the displaying of promiscuous symbol is not good yet, I have found I'm recent months not a good position to call the drawing functions. No update after some moments. But this has no problems with the radio functionality.

I draw the promiscuous symbol on the ant symbol drawing. 

Have you problems with the radio ..."cannot connect to relay" I have no idea what you mean.... Please recheck it with promiscuous mode off ... Maybe side effect from #222

Tnx for your assistance
Alex

aeickho, 2016-08-26:
Hy Michael, The radio has an 1750 beep tone now .. nice ... Tnx for reporting ... 

Please send me a link with the code plug programm for the cps 1.32...  I have only 1.30 code plugs..... 

I'll correct the frequency for release date.

Alex

aeickho, 2016-08-27:
Michael, tnx my radio has now a 1750Hz  sidebutton with ≈ 700 Hz...I'm going to search now

aeickho, 2016-08-27:
With 123a910affe60df98662cfa881fe9735559540c3 ... 1750Hz tone is working again

DL9LBG, 2016-08-27:
Great work - tnx Alex!

travisgoodspeed, 2016-08-27:
On my device with GPS, the image boots to a blank white screen, but USB is still functioning properly.  Perhaps the rtc_timer task that manages the LCD is getting stuck somewhere?

Should we add GPS support to the requirements, or close this issue and open another for bootstrapping S013.020?  (If at all possible, I'd like to merge all four of Tytera's branches back to a single firmware image that runs on all hardware.)

travisgoodspeed, 2016-08-27:
Added GPS target, and pushed core dumps to /cores/ in the repo.

aeickho, 2016-08-28:
I'll bootstrapping S013.020 this week. I think this is a straightforward job. 
... but i have no radio with gps :( so I have to work error free. 

The steps, as I imagine it
- [ ] add usb 
- [ ] Ram Dump to check memory @ DMESG_START is free
- [ ] Port other  additions

Then we can compare the too firmwares. 
Yes only one version is nicely.

-> now we talk about the S13.020 on #181

HB9TQJ, 2016-08-28:
If I try to make flash_ng it does not work:
root@inspiron6000:~/md380tools# make flash_ng
make: *** No rule to make target 'flash_ng'.  Schluss.
How can I flash the new Firmware? If I do make all flash it looks likes it uses old FW.

HB9TQJ, 2016-08-28:
I forgot to do git pull. Now it works.

travisgoodspeed, 2016-08-28:
I'm standing by with GPS hardware for testing, but a more urgent priority might be fixing #226, where D013.020 refuses to boot on GPS devices.

Alternately, we might try to get S013.020 booting on the non-GPS devices.

closed 2016-09-07 by aeickho