travisgoodspeed/md380tools #221: md380-gfx d13.020 - can't find welcomebmp

aeickho, 2016-08-25:
The SpriteStruct has "new" values

```
d02.032 
0x080f9ca8  0x002800a0 0x00010014 0x08094610 0x080fb23c

d13.020
0x080f8510  0x002800a0 0x00010014 0x20001ed4 0x080f9b3c  
                                                                                   ^--- was in flash, is now in ram

```

I think line 437 @ md380-gfx can't handle this. 

-alex

travisgoodspeed, 2016-09-08:
Perhaps we should solve this by hooking and replacing the welcome screen code, rather than patching it in-place as we currently do?  The new build process would then be to link a collection of startup logos into an ELF object, then link it into the firmware patches without modifying the firmware's Welcome logo.

That would give us more portability and the ability to choose the logo through a configuration menu, rather than at compile time.

aeickho, 2016-12-12:
I'll close it. we understand gfx_drawbmp().

closed 2016-12-12 by aeickho