travisgoodspeed/md380tools #222: D13.020 - promiscuous-mode does not work

dg9vh, 2016-08-25:
You can enable and disable promiscuous mode, but it does not enable it - it is not receiving TGs which are not in the RX-list.

I found it while talking to TG26208 on BM and not receiving me on my Tytera. 

Maybe typo again or wrong address?

Close when fixed.

aeickho, 2016-08-26:
:-(

aeickho, 2016-08-27:
I understand now, why it's not working....

dg9vh, 2016-08-27:
That's fine.

aeickho, 2016-08-27:
Hy Kim @dg9vh 
   can you please repeat the test

alex

dg9vh, 2016-08-27:
@aiecko i'll do it this evening.

dg9vh, 2016-08-27:
@aeickho tested and approved... it's working now (tested with unlisted Talkgroup, not testet with private call).

closed 2016-08-27 by dg9vh

aeickho, 2016-08-27:
private call tested be me

dg9vh, 2016-08-27:
also tested with private call to not own ID, working also... closed