travisgoodspeed/md380tools #224: MD380 Firmware Emulator

travisgoodspeed, 2016-08-27:
Howdy y'all,

I tossed together a quick emulator for the MD380 firmware, which allows the functions of the firmware to be called from the Linux command-line. It requires qemu-user and does not run on OS X or Windows.

Close this issue when the emulator has been committed to the repo along with a functional AMBE decoder. New issues will then be created for cleanup and better integration with our build process.

73 from Pizza Rat City,
--Travis

travisgoodspeed, 2016-08-27:
The initial import has been pushed.  You can run it like this to decode AMBE audio recordings on a modern Debian/Testing machine with gcc-6-arm-linux-gnueabi and qemu.

```
dell13% cd emulator
dell13% make clean all
rm -f *.o *~ md380-emu *.wav *.raw *.elf
arm-linux-gnueabi-gcc-6 -static -g    -c -o md380-emu.o md380-emu.c
arm-linux-gnueabi-objcopy \
	-I binary ../firmware/unwrapped/D002.032.img \
	--change-addresses=0x0800C000 \
	--rename-section .data=.firmware \
	-O elf32-littlearm -B arm firmware.o
arm-linux-gnueabi-objcopy \
	-I binary ../cores/d02032-core.img \
	--change-addresses=0x20000000 \
	--rename-section .data=.sram \
	-O elf32-littlearm -B arm ram.o
arm-linux-gnueabi-gcc-6 -static -g -o md380-emu md380-emu.o firmware.o ram.o \
	-Xlinker --just-symbols=../applet/src/symbols_d02.032 \
	-Xlinker --section-start=.firmware=0x0800C000 \
	-Xlinker  --section-start=.sram=0x20000000 
dell13% make play
```

travisgoodspeed, 2016-08-28:
There is now proper usage information and argument parsing.  I'll be doing a bit of cleanup and then continue to tackling the AMBE encoder.

```
dell13% ./md380-emu 
Usage: ./md380-emu [OPTION]
        -d                     Decodes AMBE
        -e                     Encodes AMBE
        -V                     Version Info
        -h                     Help!

        -v                     Verbose mode.
        -vv                    Very verbose!

        -i foo                 Input File
        -o bar                 Output File
dell13% ./md380-emu -d -i sample.amb -o output.raw 
Decoding AMBE sample.amb to 8kHz Raw Mono Signed output.raw.
Done with AMBE test.
dell13% 
```

travisgoodspeed, 2016-08-28:
Now building for D013.020, with the ambe decoder working.  (Version info is broken, but I worked around that with a weak linking symbol.)

jslatten, 2016-10-08:
Is it possible to use the AMBE encoder/decoder code to create DSTAR/P25/DMR radio using software instead of hardware chip?

travisgoodspeed, 2016-10-08:
Yes, the decoder is working and already committed.

The encoder will work, whenever I find the time to finish it.

ispkorte, 2016-10-09:
Thanks Travis
i use **Digital Speech Decoder** on **Linux Mint 17.3** ; works fine with a SDR (gqrx)

travisgoodspeed, 2016-12-12:
@aeickho Are we good to close this issue now?

aeickho, 2016-12-12:
yes ;)

closed 2016-12-12 by aeickho

travisgoodspeed, 2016-12-12:
Woohoo!