travisgoodspeed/md380tools #225: make: *** [flashdb] Error 1

ecarvalho123, 2016-08-28:
Running this on Windows 10
Attempting to perform: make flashdb

Edward@C-R3ACTION MINGW64 ~/Documents/git/travisgoodspeed/md380tools (master)
$ make flashdb
cd db && C:/GnuWin32/bin/make
make[1]: Entering directory `C:/Users/Edward/Documents/git/travisgoodspeed/md380tools/db'
make[1]: Nothing to be done for `update'.
make[1]: Leaving directory `C:/Users/Edward/Documents/git/travisgoodspeed/md380tools/db'
iconv -c -f UTF-8 -t ascii//TRANSLIT db/users.csv | cut -d',' -f1-3,5-6 | sed 's/,\s+/,/g' > data.csv
wc -c < data.csv > data
cat data.csv >> data
./md380-tool spiflashwrite data 0x100000
**No backend available
make: *** [flashdb] Error 1**

---
Please advise

travisgoodspeed, 2016-08-28:
It looks like you haven't got libusb installed, but you don't have whatever driver it needs to connect to devices on Windows.

Cheers,
--Travis

(Close this issue if a fix is describes, or if conversation goes quiet for two days.)

M0IVQ, 2016-08-28:
Yes that is the error message you get when either libusb is missing or the driver hasn't been configured. Instructions for the driver are in windows-instructions.txt

Make flashdb doesn't do anything with gnuwin32 for some reason (goes through the motions but doesn't make any change to the radio) but that is being tracked in another issue.

travisgoodspeed, 2016-08-28:
Okay, closing this as a duplicate of #186.

closed 2016-08-28 by travisgoodspeed

ecarvalho123, 2016-08-29:
The error
No backend available
make: *** [flashdb] Error 1

... was indeed caused by not having libusb installed. The install steps were actually followed however, Windows 10 was not allowing for unassigned drivers to be installed which gave me a false impression of success but questions raised when windows alerted of unassigned drivers were blocked...
Once that functionality was disabled, I successfully installed libusb drivers and the error is gone...
Still there is the issue of not overwriting the radio memory. No change after make flashdb and the green dialog box still says userid.csv needs to be installed.... 

I will follow #186 
Thank you

E