travisgoodspeed/md380tools #226: MD390 GPS - White screen only after welcome screen

oh2lak, 2016-08-28:
Flashing MD390 GPS with the md380tools results in a white screen, no channels visible, nor the radio won't transmit or receive. The bootup is OK, get the rolling welcome screen and my ID displayed, but only a white screen after that. Shutdown reveals the normal display during the power-off also.

I can read and program the radio also without any problem, but the radio is unusable.

Reflashing back to factory firmware went also OK.

md380-tool dmesg;
Starting main()
Performing AES startup check.
aes_loadkey (0x20014a80):  f5 c2 f1 8f 63 52 f5 a6 f0 6a a5 ed 5e 9c df 14
aes_cipher(0x20014a60);
IN  : 58 e3 1d 17 42 00 47 00 0f 51 35 32 38 33 31 35
OUT : b7 5a 0c 25 67 61 1a 26 52 d0 32 99 5e d1 2b 7a
Startup AES check failed.  Attempting to forge the results.
*0x2001d39b = 0x00

md380-tool spiflashid;
SPI Flash ID: ef 40 18
W25Q128FV 16MByte

Any help?

travisgoodspeed, 2016-08-28:
This is a bug that is new to the D002.032 firmware, which was recently made default for our patching. See #218 for details.

For now, you can simply build the older firmware for your radio.  It works fine in models with GPS, but does not use the GPS.

`make flash_d02.032`

Astralix, 2016-08-30:
Just want to comment, that I have the same problem with MD-380G (380 new version with GPS).

`make flash_d02.032` 
reverts the firmware to an older but working state.

aeickho, 2016-08-30:
@astralix .. with which fw version the radio was delivered?

Astralix, 2016-08-30:
Wow... I unpacked and reflashed it with your firmware... 
According the firmware ZIP I downloaded at the day of delivery, it was 
"Firmware 13.20 for MD-390 with GPS  and MD-380G with GPS and new vocoder"
containing an image "TYT-Vocoder-MD390-S13.20.bin"

javastraat, 2016-09-02:
Same problem here:
New MD390-GPS default Firmware was S13.020
Original firmware S13.020 no problem
Patched firmware 2.032  no problem
Patched D13.020 boots with welcome screen and power off screen

thanx for all the work

bendikz, 2016-09-05:
Can boot with patched S13.020 firmware, no functionality other than yet but it works! 
[image]

bendikz, 2016-09-05:
See #181. I believe this issue is more related to S013.020 as that is the GPS-specific, therefore the D (non-GPS) version **won't work**. Patched S013.020 works, however it's not done yet. Stick with 2.032 for now.

travisgoodspeed, 2016-09-08:
Closing this issue as a (semi) duplicate of #181.

closed 2016-09-08 by travisgoodspeed