travisgoodspeed/md380tools #227: MD380 (old vocoder) d13.020 - modified roger beep not working

dg9vh, 2016-08-28:
Call me bug-hunter, hi...

Roger beep modification is not working with actual build of firmware based on d13.020.

Menuitem saves state but does not modify beep.

close when killed

travisgoodspeed, 2016-08-28:
Assigning @aeickho as this is probably a dupe of #218.

aeickho, 2016-08-30:
@dg9vh please enable 
global_addl_config.debug <- Debug enable

and watch with ./md380-tool dmesgtail for the number of beep tone 

OSMboxPend @ 8031181, 0    <- the 0 ist the type of beeptone

temporary only   roger beep and dmr sync are mortified, not all similar to d02.32

dg9vh, 2016-08-30:
@aeickho I'll test it this evening.

dg9vh, 2016-08-30:
@aeickho it is OSMboxPend @ 8031181, 9  but I get no modified tone on every beep the thing produces....

aeickho, 2016-08-30:
fixed with 8485488c29398048b012f9bda76808df4a03984a

closed 2016-08-30 by aeickho

dg9vh, 2016-08-30:
Yesssss!