travisgoodspeed/md380tools #228: SPI Flash ID: 70 f1 1

yoony64, 2016-08-28:
Hi,

My MD-380 has a SPI Flash ID of 70 f1 1 and it won't flash the users.csv.  This radio had the D013.020 firmware(patched).

Help!

Thank you.

travisgoodspeed, 2016-08-28:
On what operating system are you trying to flash it?  We have a known issue that it doesn't work in MINGW, issue #186.

yoony64, 2016-08-28:
I've been using Windows 7 64 bit.

travisgoodspeed, 2016-08-28:
Then see issue #186, which probably describes what you're seeing.

Closing this issue as a duplicate.

closed 2016-08-28 by travisgoodspeed

yoony64, 2016-08-28:
My radio was originally a V13.20 firmware, before modifying.  I mistakenly used a V2.32 based patch firmware and everything seems to work, but the users.csv won't work.  

With this wrong firmware, the make flashdb thinks the SPI is 10 dc 1.

Once I corrected the firmware to the patched 13.20 version, now the SPI ID says 70 f1 1 and make flashdb fails with unknown ID.