travisgoodspeed/md380tools #231: 'make flash' fails on fresh system

M0IVQ, 2016-08-29:
If I run 'make flash' on a clean system, I get the following error:

> curl -f -L $(cat firmware_files.txt | grep "^D013.020.zip" | cut -f2) -o dl/D013                                                                                                    .020.zip
> curl: no URL specified!

If I then run 'make all', build completes without errors. I can then run 'make [clean] flash' and all works fine as firmware files have already downloaded.

This occurs on both linux and Cygwin.

M0IVQ, 2016-08-29:
changing the line:

`flash:`
to
`flash: applets`

in the main makefile seems to make it work

IU3CUJ, 2016-08-30:
I want to start by saying that I am very new to linux and DMR procedures. The only way in which I managed to do to upgrade my RT3 was with loading from the Windows BIN, I always failed by my Mint Linux-based Ubuntu.
I tried in every forum but the procedures always give me the same result

(...)
make[2]: uscita dalla directory "/home/fabrizio/md380tools/applet/lib"
arm-none-eabi-gcc -std=gnu99 -g -O2 -Wall -Tapplet_d13.020.ld -mlittle-endian -mthumb -mcpu=cortex-m4 -mthumb-interwork -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fshort-wchar -DGIT_VERSION=\"c82c-dirty\" -Iinc -Ilib -Ilib/inc -I. -Ilib/inc/core -Ilib/inc/peripherals -D "MD380_d13_020"     -c -o main.o src/main.c
In file included from lib/inc/core/core_cm4.h:143:0,
                 from lib/inc/stm32f4xx.h:245,
                 from lib/inc/stm32f4_discovery.h:32,
                 from src/main.c:7:
/usr/lib/gcc/arm-none-eabi/4.8.2/include/stdint.h:9:26: fatal error: stdint.h: No such file or directory
 # include_next <stdint.h>
                          ^
compilation terminated.
make[1]: *** [main.o] Errore 1
make[1]: uscita dalla directory "/home/fabrizio/md380tools/applet"
make: *** [applets] Errore 2

Of course, after this failure the handset did not appear changed. So I opted to upload the files from Windows (Sigh!) And then give the flashdb sudo command to update the csv, and this does work properly.

I do not know where I'm wrong ...
Can someone help me ?

sijskes, 2016-08-30:
You could try with
```
sudo apt-get install libnewlib-arm-none-eabi
```

aeickho, 2016-08-31:
Hi Simon,
   Yes.... I think ... Travis-cv is not calling make flash

I'll fix it today

IU3CUJ, 2016-09-01:
Great sijskes !! Work fine now !

sijskes, 2016-09-01:
@aeickho 2 different problems in 1 issue. @IU3CUJ issue is fixed and can be closed. If @m0ivq issue is fixed, you can close issue.

travisgoodspeed, 2016-09-14:
Looks like this bug is fixed.  Closing the issue.

m0ivq, 2016-10-02:
Still fails unless 'make all' has already been run on fresh git clone:

```
cd ../../firmware && C:/Progra~1/GnuWin32/bin/make unwrapped/D013.020.img
make[3]: Entering directory `C:/Users/post2/Desktop/git/travisgoodspeed/md380tools/firmware'
which curl >>/dev/null #Check that we have curl.
curl -f -L $(cat firmware_files.txt | grep "^D013.020.zip" | cut -f2) -o dl/D013.020.zip
curl: no URL specified!
curl: try 'curl --help' or 'curl --manual' for more information
make[3]: *** [dl/D013.020.zip] Error 2
make[3]: Leaving directory `C:/Users/post2/Desktop/git/travisgoodspeed/md380tools/firmware'
make[2]: *** [unwrapped.img] Error 2
make[2]: Leaving directory `C:/Users/post2/Desktop/git/travisgoodspeed/md380tools/patches/d13.020'
make[1]: *** [experiment.img] Error 2
make[1]: Leaving directory `C:/Users/post2/Desktop/git/travisgoodspeed/md380tools/applet'
make: *** [flash] Error 2
```
looking at firmware_files.txt, there is no line with "D013.020.zip" (only D013.020.bin)

sijskes, 2016-11-01:
refactored make flash target with dependencies. closing.

closed 2016-11-01 by sijskes