travisgoodspeed/md380tools #233: Unkown SPI Flash ID: 70 f1 1 (MD-380, old vocoder, experimental D13.020)

burnersk, 2016-08-30:
Flashing (`make all flash`) experimental D13.020 (from e2ea7e85fc651a10a618dfa2887d997c75f771e4) the SPI Flash ID of my MD-380 (old vocoder) seems to be changed.

Also the USB device label have changed from `Digital Radio in USB mode` to `Patched MD380`. I also reinstalled the new drivers (libusb inf-wizzard).

```
$ ./md380-tool spiflashid
SPI Flash ID: 70 f1 1
Unkown SPI Flash - please report
```

```
$ make flashdb
cd db && C:/Progra~2/GnuWin32/bin/make
make[1]: Entering directory `C:/Users/soeren/Documents/md380tools/db'
make[1]: F▒r das Ziel ▒update▒ ist nichts zu tun.
make[1]: Leaving directory `C:/Users/soeren/Documents/md380tools/db'
iconv -c -f UTF-8 -t ascii//TRANSLIT db/users.csv | cut -d',' -f1-3,5-6 | sed 's/,\s+/,/g' > data.csv
wc -c < data.csv > data
cat data.csv >> data
./md380-tool spiflashwrite data 0x100000
SPI Flash ID: 70 f1 1
Unkown SPI Flash - please report
can't programm spi flash wrong flash type
```

is there any quick 'n dirty fix to upload the user database?

M0IVQ, 2016-08-30:
see #186

travisgoodspeed, 2016-08-30:
Closing issue as a duplicate of #186.

closed 2016-08-30 by travisgoodspeed

yashm, 2017-04-22:
I am also getting this error.
@burnersk which fix worked best for you?