travisgoodspeed/md380tools #236: screen fixed when signal received

sijskes, 2016-08-30:
When a signal is received, the menu is not displayed by pressing a button.
Volume knob is functional. No volume screen is displayed.
When screensaver kicks in during qso key activation displays screen, but no menu activated/displayed.
Volume knob does not reactivate display.

mach327, 2016-08-30:
Regarding the menu during receive: reference #60

sijskes, 2016-08-31:
@mach327 roger. ideally the menu should appear when toprow buttons are pushed. or only the green ne.
i think the program flow in the original fw became a lot harder, because of the animated scrolling of text and 'eye' indicator in top of screen. i think the screen is redrawn more than in prev versions.

aeickho, 2016-08-31:
I close this issue now :)

The menu / display handling is nearly completely not understand yet. 

Bug reports makes not so much sense ... LoL
We can fill big lists... 

I have spent more then 10 hours to find an good hook location to display the "eye" ... This is the product ... An maybe "eye"

closed 2016-08-31 by aeickho