travisgoodspeed/md380tools #237: TYT MD-390 no GPS D13.020 - Windows - mirorred and flipped display ...

Conrad007, 2016-08-31:
Hello,
I have flashed the experimental firmware v1 and later also v2 (from 22.April) using the original upgrade tool, but after that I just get a horizontal flipped and mirrored welcome screen (showing my callsign and name) on white background and the device reboots into this every 20 seconds... Fortunately, I was able to bring it back to normal life by flashing again the original firmware D13.020, but I would like to have md380tools ...

aeickho, 2016-08-31:
Yes,
   The fw from 22.april has not md380 support.

Checkout the actual sources ... And build your own experimental fw.

... Ask for support to build the fw on md380   group on Google ... See bottom of readme.md

This is not an support forum ..  ;)

73 Alex df8av

aeickho, 2016-08-31:
#178

closed 2016-08-31 by aeickho

Conrad007, 2016-08-31:
Ok, thanks Alex and sorry for posting in the wrong place ...
By the way, I have an **md390, new vocoder, no GPS** and I tried also the older fw (experimental) with the same results... (only for info).
Generally, should I quit trying on Windows and build my exp. fw in Linux?

73 Conrad

Conrad007, 2016-09-01:
Solved and working! I was just using my Raspberry Pi with Debian Jessie. I'm very pleased, thanks for all you nice work!