travisgoodspeed/md380tools #238: personal development workflow documentation

sijskes, 2016-08-31:
It would be nice to see some notes on the development workflow. 
- is testing possible without disconnect and reboot? (sound?)
- does it make sense to run in qemu?

aeickho, 2016-08-31:
Hy Simon,
      My workflow...  Reading the fw with radare2 writing annotations. If I find an interesting place /  function. I write an diagnostic hook to see the calling parameters via md380tools dmsgtail .

My start point was the stm32 internals @0x40000000 ... Clock, spi, gpio ... 

This was the way to find the date format. :)

I think qemu-arm is not for all a good support. There are so many. Of tasks .. and internal communications... This is not easy to simulate. 

I like debug with printf ;)

And there is a lot of debug log space free on spi flash 

Also .. an good start point create task

73 from Goettingen, 
     Alex df8av

aeickho, 2016-08-31:
[..]

>- is testing possible without disconnect and reboot? (sound?)

I have solder a cable .. that's allowed USB and sound

aeickho, 2016-09-01:
if you @sijskes have no more questions please close this issue, tnx

sijskes, 2016-09-01:
i think we should let this issue open until we have created a document from the information in this issue.
it does not have to happen today or tomorrow.

travisgoodspeed, 2016-09-08:
Agreed.

Assigning myself, @aeickho and @cr to document our workflow.  Unassign yourself as documentation for your workflow is written.

travisgoodspeed, 2016-09-08:
Opened #269 to track the creation of the wiki.  I'll write tutorials for opening the firmware in IDA Pro and Radare2 when I get a chance.

closed 2017-01-21 by travisgoodspeed