travisgoodspeed/md380tools #239: Feature request : An instructional Youtube video of how to get latest github fw and update the radio.

geiraage, 2016-08-31:
Hello Travis and other contributors to this project.

This is probably one of the most interesting projects for DMR enthusiasts at the moment. And I know there is a quite good windows-instructions.txt in the root directory. 

I was wondering if someone could make an instructional video from start to end how to get the latest version from Github? With that I mean

How to Install needed windows software.
How to clone current repository. 
How to update firmware from that updated repository to the radio. 

I think this will boost the interest of using firmware. Especially for us noobs. I know this is not actually a feature nor issue, but still I think its a good idea. 

If you do a search on YouTube, you will find 2 videos of interest.

How to update TYT MD-380 normal or hacked firmware! (1 yr old)
https://www.youtube.com/watch?v=IfBw87kpuYs

This video is using the official firmware ver. 2.0 from 22 of april 2016. Found different places on the net. 

And a presentation of the MD380Tool firmware 
TYT MD-380 Firmware Saar Edition 
https://www.youtube.com/watch?v=c6S8LHh1v7g

aeickho, 2016-08-31:
I add @ladyada to this issue....  ;)

Maybe she has fun to make a new video. 

I can't help with supporting windows installation, my knowledge is not enough.

.And also we have no automatic continually testing like travis-cv for windows. 

From Göttingen
Alex df8av

ladyada, 2016-08-31:
hiya - ive found for such complex work, a video isn't always best (and of course can't be udpated as urls and other stuff changes)

check out my written guides
https://learn.adafruit.com/tytera-md-380-dmr/
which ought to be up to date and
https://learn.adafruit.com/windows-tools-for-the-electrical-engineer
for the tools

anontuta, 2016-09-01:
If posible record video for android md380 tools.

Thank you.

travisgoodspeed, 2016-09-01:
The android tools are not yet ready for public use.  I'll do an open beta when they are, but until then, only developers should mess with it.

closed 2016-10-17 by travisgoodspeed