travisgoodspeed/md380tools #254: M.RogerBeep dont work on my MD-380.

PD1LOI, 2016-09-04:
It is known that M.Roger Beep does not work as well as in the older firmware?
In seems to work normally, Roger Beep on my MD-380.

travisgoodspeed, 2016-09-05:
We need a bit more information to track down the issue.  Could you please describe exactly which version you are now running, and what the Roger Beep does instead?

PD1LOI, 2016-09-06:
cd patches/2.032 && make clean
make[1]: Entering directory '/home/zarya/md380tools/patches/2.032'
rm -f *~ *.bin *.img
make[1]: Leaving directory '/home/zarya/md380tools/patches/2.032'
cd patches/d13.020 && make clean
make[1]: Entering directory '/home/zarya/md380tools/patches/d13.020'
rm -f *~ *.bin *.img
make[1]: Leaving directory '/home/zarya/md380tools/patches/d13.020'
cd firmware && make clean
make[1]: Entering directory '/home/zarya/md380tools/firmware'
rm -rf bin unwrapped
make[1]: Leaving directory '/home/zarya/md380tools/firmware'
cd applet && make clean
make[1]: Entering directory '/home/zarya/md380tools/applet'
make -C lib clean
make[2]: Entering directory '/home/zarya/md380tools/applet/lib'
rm -f misc.o stm32f4xx_dma.o stm32f4xx_rcc.o stm32f4xx_adc.o stm32f4xx_exti.o stm32f4xx_rng.o stm32f4xx_can.o stm32f4xx_flash.o stm32f4xx_rtc.o stm32f4xx_crc.o stm32f4xx_fsmc.o stm32f4xx_sdio.o stm32f4xx_cryp_aes.o stm32f4xx_gpio.o stm32f4xx_spi.o stm32f4xx_cryp.o stm32f4xx_hash.o stm32f4xx_syscfg.o stm32f4xx_cryp_des.o stm32f4xx_hash_md5.o stm32f4xx_tim.o stm32f4xx_cryp_tdes.o stm32f4xx_hash_sha1.o stm32f4xx_usart.o stm32f4xx_dac.o stm32f4xx_i2c.o stm32f4xx_wwdg.o stm32f4xx_dbgmcu.o stm32f4xx_iwdg.o stm32f4xx_dcmi.o stm32f4xx_pwr.o libstm32f4.a
make[2]: Leaving directory '/home/zarya/md380tools/applet/lib'
rm -f *.img *.bin *.elf *.hex *.o *~ *.sym src/*~ src/version.h
make[1]: Leaving directory '/home/zarya/md380tools/applet'
rm -f *~ *.pyc
cd firmware && make all
make[1]: Entering directory '/home/zarya/md380tools/firmware'
mkdir -p bin
unzip -p dl/D002.024.zip "$(cat firmware_files.txt | grep "^D002.024" | cut -f3)" >bin/D002.024.bin
unzip -p dl/D002.025.zip "$(cat firmware_files.txt | grep "^D002.025" | cut -f3)" >bin/D002.025.bin
unzip -p dl/D002.026.zip "$(cat firmware_files.txt | grep "^D002.026" | cut -f3)" >bin/D002.026.bin
unzip -p dl/D002.027.zip "$(cat firmware_files.txt | grep "^D002.027" | cut -f3)" >bin/D002.027.bin
unzip -p dl/D002.030.zip "$(cat firmware_files.txt | grep "^D002.030" | cut -f3)" >bin/D002.030.bin
unzip -p dl/D002.032.zip "$(cat firmware_files.txt | grep "^D002.032" | cut -f3)" >bin/D002.032.bin
unzip -p dl/D002.034.zip "$(cat firmware_files.txt | grep "^D002.034" | cut -f3)" >bin/D002.034.bin
unzip -p dl/D003.008.zip "$(cat firmware_files.txt | grep "^D003.008" | cut -f3)" >bin/D003.008.bin
cp dl/D003.020.bin bin/D003.020.bin
cp dl/S003.012.bin bin/S003.012.bin
cp dl/D013.009.bin bin/D013.009.bin
cp dl/D013.014.bin bin/D013.014.bin
cp dl/D013.020.bin bin/D013.020.bin
cp dl/S013.012.bin bin/S013.012.bin
cp dl/S013.020.bin bin/S013.020.bin
cp dl/MD280.002.018.bin bin/MD280.002.018.bin
cp dl/MD446.D003.020.bin bin/MD446.D003.020.bin
cp dl/MD446.D013.020.bin bin/MD446.D013.020.bin
mkdir -p unwrapped
../md380-fw --unwrap bin/D002.024.bin unwrapped/D002.024.img
DEBUG: reading "bin/D002.024.bin"
INFO: base address 0x800c000
INFO: length 0xf4000
DEBUG: writing "unwrapped/D002.024.img"
../md380-fw --unwrap bin/D002.025.bin unwrapped/D002.025.img
DEBUG: reading "bin/D002.025.bin"
INFO: base address 0x800c000
INFO: length 0xf2800
DEBUG: writing "unwrapped/D002.025.img"
../md380-fw --unwrap bin/D002.026.bin unwrapped/D002.026.img
DEBUG: reading "bin/D002.026.bin"
INFO: base address 0x800c000
INFO: length 0xf2800
DEBUG: writing "unwrapped/D002.026.img"
../md380-fw --unwrap bin/D002.027.bin unwrapped/D002.027.img
DEBUG: reading "bin/D002.027.bin"
INFO: base address 0x800c000
INFO: length 0xf2800
DEBUG: writing "unwrapped/D002.027.img"
../md380-fw --unwrap bin/D002.030.bin unwrapped/D002.030.img
DEBUG: reading "bin/D002.030.bin"
INFO: base address 0x800c000
INFO: length 0xf2c00
DEBUG: writing "unwrapped/D002.030.img"
../md380-fw --unwrap bin/D002.032.bin unwrapped/D002.032.img
DEBUG: reading "bin/D002.032.bin"
[... 431 lines cut ...]
/usr/lib/gcc/arm-none-eabi/4.9/../../../arm-none-eabi/bin/ld: warning: lib/libstm32f4.a(stm32f4xx_gpio.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.9/../../../arm-none-eabi/bin/ld: warning: lib/libstm32f4.a(stm32f4xx_rcc.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
arm-none-eabi-objcopy -O binary main.elf main.img
arm-none-eabi-objdump -t main.elf >main.img.sym
cd ../patches/d13.020 && make all
make[2]: Entering directory '/home/zarya/md380tools/patches/d13.020'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/zarya/md380tools/patches/d13.020'
cp ../patches/d13.020/patched.img ./experiment.img
# use a fresh experiment.img
cd ../patches/d13.020 && make all
make[2]: Entering directory '/home/zarya/md380tools/patches/d13.020'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/zarya/md380tools/patches/d13.020'
cp ../patches/d13.020/patched.img ./experiment.img
python2 merge_d13.020.py experiment.img main.img 0x0809b000
Merging an applet.
Loading symbols from main.img.sym
Merging main.img into experiment.img at 0809b000
../md380-fw --wrap experiment.img experiment.bin
DEBUG: reading "experiment.img"
INFO: base address 0x800c000
INFO: length 0xf3000
DEBUG: writing "experiment.bin"
make[1]: Leaving directory '/home/zarya/md380tools/applet'

aeickho, 2016-09-07:
@PD1LOI the build process  is correct. why do you think it is wrong? 

please describe your problem with "rogerbeep"

PD1LOI, 2016-09-07:
Is there a difference in M.Roger Beep? Here the firmware where does show are low.

cd patches/2.032 && make clean
make[1]: Entering directory '/home/zarya/md380tools/patches/2.032'
rm -f *~ *.bin *.img
make[1]: Leaving directory '/home/zarya/md380tools/patches/2.032'
cd patches/d13.020 && make clean
make[1]: Entering directory '/home/zarya/md380tools/patches/d13.020'
rm -f *~ *.bin *.img
make[1]: Leaving directory '/home/zarya/md380tools/patches/d13.020'
cd firmware && make clean
make[1]: Entering directory '/home/zarya/md380tools/firmware'
rm -rf bin unwrapped
make[1]: Leaving directory '/home/zarya/md380tools/firmware'
cd applet && make clean
make[1]: Entering directory '/home/zarya/md380tools/applet'
make -C lib clean
make[2]: Entering directory '/home/zarya/md380tools/applet/lib'
rm -f misc.o stm32f4xx_dma.o stm32f4xx_rcc.o stm32f4xx_adc.o stm32f4xx_exti.o stm32f4xx_rng.o stm32f4xx_can.o stm32f4xx_flash.o stm32f4xx_rtc.o stm32f4xx_crc.o stm32f4xx_fsmc.o stm32f4xx_sdio.o stm32f4xx_cryp_aes.o stm32f4xx_gpio.o stm32f4xx_spi.o stm32f4xx_cryp.o stm32f4xx_hash.o stm32f4xx_syscfg.o stm32f4xx_cryp_des.o stm32f4xx_hash_md5.o stm32f4xx_tim.o stm32f4xx_cryp_tdes.o stm32f4xx_hash_sha1.o stm32f4xx_usart.o stm32f4xx_dac.o stm32f4xx_i2c.o stm32f4xx_wwdg.o stm32f4xx_dbgmcu.o stm32f4xx_iwdg.o stm32f4xx_dcmi.o stm32f4xx_pwr.o libstm32f4.a
make[2]: Leaving directory '/home/zarya/md380tools/applet/lib'
rm -f *.img *.bin *.elf *.hex *.o *~ *.sym src/*~ src/version.h
make[1]: Leaving directory '/home/zarya/md380tools/applet'
rm -f *~ *.pyc
cd firmware && make all
make[1]: Entering directory '/home/zarya/md380tools/firmware'
mkdir -p bin
unzip -p dl/D002.024.zip "$(cat firmware_files.txt | grep "^D002.024" | cut -f3)" >bin/D002.024.bin
unzip -p dl/D002.025.zip "$(cat firmware_files.txt | grep "^D002.025" | cut -f3)" >bin/D002.025.bin
unzip -p dl/D002.026.zip "$(cat firmware_files.txt | grep "^D002.026" | cut -f3)" >bin/D002.026.bin
unzip -p dl/D002.027.zip "$(cat firmware_files.txt | grep "^D002.027" | cut -f3)" >bin/D002.027.bin
unzip -p dl/D002.030.zip "$(cat firmware_files.txt | grep "^D002.030" | cut -f3)" >bin/D002.030.bin
unzip -p dl/D002.032.zip "$(cat firmware_files.txt | grep "^D002.032" | cut -f3)" >bin/D002.032.bin
unzip -p dl/D002.034.zip "$(cat firmware_files.txt | grep "^D002.034" | cut -f3)" >bin/D002.034.bin
unzip -p dl/D003.008.zip "$(cat firmware_files.txt | grep "^D003.008" | cut -f3)" >bin/D003.008.bin
cp dl/D003.020.bin bin/D003.020.bin
cp dl/S003.012.bin bin/S003.012.bin
cp dl/D013.009.bin bin/D013.009.bin
cp dl/D013.014.bin bin/D013.014.bin
cp dl/D013.020.bin bin/D013.020.bin
cp dl/S013.012.bin bin/S013.012.bin
cp dl/S013.020.bin bin/S013.020.bin
cp dl/MD280.002.018.bin bin/MD280.002.018.bin
cp dl/MD446.D003.020.bin bin/MD446.D003.020.bin
cp dl/MD446.D013.020.bin bin/MD446.D013.020.bin
mkdir -p unwrapped
../md380-fw --unwrap bin/D002.024.bin unwrapped/D002.024.img
DEBUG: reading "bin/D002.024.bin"
INFO: base address 0x800c000
INFO: length 0xf4000
DEBUG: writing "unwrapped/D002.024.img"
../md380-fw --unwrap bin/D002.025.bin unwrapped/D002.025.img
DEBUG: reading "bin/D002.025.bin"
INFO: base address 0x800c000
INFO: length 0xf2800
DEBUG: writing "unwrapped/D002.025.img"
../md380-fw --unwrap bin/D002.026.bin unwrapped/D002.026.img
DEBUG: reading "bin/D002.026.bin"
INFO: base address 0x800c000
INFO: length 0xf2800
DEBUG: writing "unwrapped/D002.026.img"
../md380-fw --unwrap bin/D002.027.bin unwrapped/D002.027.img
DEBUG: reading "bin/D002.027.bin"
INFO: base address 0x800c000
INFO: length 0xf2800
DEBUG: writing "unwrapped/D002.027.img"
../md380-fw --unwrap bin/D002.030.bin unwrapped/D002.030.img
DEBUG: reading "bin/D002.030.bin"
INFO: base address 0x800c000
INFO: length 0xf2c00
DEBUG: writing "unwrapped/D002.030.img"
[... 4937 lines cut ...]
/usr/lib/gcc/arm-none-eabi/4.9/../../../arm-none-eabi/bin/ld: warning: rtc_timer.o uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.9/../../../arm-none-eabi/bin/ld: warning: i2c.o uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.9/../../../arm-none-eabi/bin/ld: warning: display.o uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.9/../../../arm-none-eabi/bin/ld: warning: lib/libstm32f4.a(stm32f4xx_gpio.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.9/../../../arm-none-eabi/bin/ld: warning: lib/libstm32f4.a(stm32f4xx_rcc.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
arm-none-eabi-objcopy -O binary main.elf main.img
arm-none-eabi-objdump -t main.elf >main.img.sym
cd ../patches/2.032 && make all
make[2]: Entering directory '/home/zarya/md380tools/patches/2.032'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/zarya/md380tools/patches/2.032'
cp ../patches/2.032/patched.img ./experiment.img
# use a fresh experiment.img
cp ../patches/2.032/patched.img ./experiment.img
python2 merge_d02.032.py experiment.img main.img 0x0809D000
Merging an applet.
Loading symbols from main.img.sym
Hooking a menu call.
Merging main.img into experiment.img at 0809d000
../md380-fw --wrap experiment.img experiment.bin
DEBUG: reading "experiment.img"
INFO: base address 0x800c000
INFO: length 0xf2c00
DEBUG: writing "experiment.bin"
make[1]: Leaving directory '/home/zarya/md380tools/applet'

PD1LOI, 2016-09-07:
This firmware 20160825-190001-660ef4e6.bin has a low tone rogerbeeb and control tones.
and that the latest firmware is not
Here the firmware txt

cd patches/2.032 && make clean
make[1]: Entering directory '/home/zarya/md380tools/patches/2.032'
rm -f *~ *.bin *.img
make[1]: Leaving directory '/home/zarya/md380tools/patches/2.032'
cd patches/d13.020 && make clean
make[1]: Entering directory '/home/zarya/md380tools/patches/d13.020'
rm -f *~ *.bin *.img
make[1]: Leaving directory '/home/zarya/md380tools/patches/d13.020'
cd firmware && make clean
make[1]: Entering directory '/home/zarya/md380tools/firmware'
rm -rf bin unwrapped
make[1]: Leaving directory '/home/zarya/md380tools/firmware'
cd applet && make clean
make[1]: Entering directory '/home/zarya/md380tools/applet'
make -C lib clean
make[2]: Entering directory '/home/zarya/md380tools/applet/lib'
rm -f misc.o stm32f4xx_dma.o stm32f4xx_rcc.o stm32f4xx_adc.o stm32f4xx_exti.o stm32f4xx_rng.o stm32f4xx_can.o stm32f4xx_flash.o stm32f4xx_rtc.o stm32f4xx_crc.o stm32f4xx_fsmc.o stm32f4xx_sdio.o stm32f4xx_cryp_aes.o stm32f4xx_gpio.o stm32f4xx_spi.o stm32f4xx_cryp.o stm32f4xx_hash.o stm32f4xx_syscfg.o stm32f4xx_cryp_des.o stm32f4xx_hash_md5.o stm32f4xx_tim.o stm32f4xx_cryp_tdes.o stm32f4xx_hash_sha1.o stm32f4xx_usart.o stm32f4xx_dac.o stm32f4xx_i2c.o stm32f4xx_wwdg.o stm32f4xx_dbgmcu.o stm32f4xx_iwdg.o stm32f4xx_dcmi.o stm32f4xx_pwr.o libstm32f4.a
make[2]: Leaving directory '/home/zarya/md380tools/applet/lib'
rm -f *.img *.bin *.elf *.hex *.o *~ *.sym src/*~ src/version.h
make[1]: Leaving directory '/home/zarya/md380tools/applet'
rm -f *~ *.pyc
cd firmware && make all
make[1]: Entering directory '/home/zarya/md380tools/firmware'
mkdir -p bin
unzip -p dl/D002.024.zip "$(cat firmware_files.txt | grep "^D002.024" | cut -f3)" >bin/D002.024.bin
unzip -p dl/D002.025.zip "$(cat firmware_files.txt | grep "^D002.025" | cut -f3)" >bin/D002.025.bin
unzip -p dl/D002.026.zip "$(cat firmware_files.txt | grep "^D002.026" | cut -f3)" >bin/D002.026.bin
unzip -p dl/D002.027.zip "$(cat firmware_files.txt | grep "^D002.027" | cut -f3)" >bin/D002.027.bin
unzip -p dl/D002.030.zip "$(cat firmware_files.txt | grep "^D002.030" | cut -f3)" >bin/D002.030.bin
unzip -p dl/D002.032.zip "$(cat firmware_files.txt | grep "^D002.032" | cut -f3)" >bin/D002.032.bin
unzip -p dl/D002.034.zip "$(cat firmware_files.txt | grep "^D002.034" | cut -f3)" >bin/D002.034.bin
unzip -p dl/D003.008.zip "$(cat firmware_files.txt | grep "^D003.008" | cut -f3)" >bin/D003.008.bin
cp dl/D003.020.bin bin/D003.020.bin
cp dl/S003.012.bin bin/S003.012.bin
cp dl/D013.009.bin bin/D013.009.bin
cp dl/D013.014.bin bin/D013.014.bin
cp dl/D013.020.bin bin/D013.020.bin
cp dl/S013.012.bin bin/S013.012.bin
cp dl/S013.020.bin bin/S013.020.bin
cp dl/MD280.002.018.bin bin/MD280.002.018.bin
cp dl/MD446.D003.020.bin bin/MD446.D003.020.bin
cp dl/MD446.D013.020.bin bin/MD446.D013.020.bin
mkdir -p unwrapped
../md380-fw --unwrap bin/D002.024.bin unwrapped/D002.024.img
DEBUG: reading "bin/D002.024.bin"
INFO: base address 0x800c000
INFO: length 0xf4000
DEBUG: writing "unwrapped/D002.024.img"
../md380-fw --unwrap bin/D002.025.bin unwrapped/D002.025.img
DEBUG: reading "bin/D002.025.bin"
INFO: base address 0x800c000
INFO: length 0xf2800
DEBUG: writing "unwrapped/D002.025.img"
../md380-fw --unwrap bin/D002.026.bin unwrapped/D002.026.img
DEBUG: reading "bin/D002.026.bin"
INFO: base address 0x800c000
INFO: length 0xf2800
DEBUG: writing "unwrapped/D002.026.img"
../md380-fw --unwrap bin/D002.027.bin unwrapped/D002.027.img
DEBUG: reading "bin/D002.027.bin"
INFO: base address 0x800c000
INFO: length 0xf2800
DEBUG: writing "unwrapped/D002.027.img"
../md380-fw --unwrap bin/D002.030.bin unwrapped/D002.030.img
DEBUG: reading "bin/D002.030.bin"
INFO: base address 0x800c000
[... 4939 lines cut ...]
/usr/lib/gcc/arm-none-eabi/4.9/../../../arm-none-eabi/bin/ld: warning: rtc_timer.o uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.9/../../../arm-none-eabi/bin/ld: warning: i2c.o uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.9/../../../arm-none-eabi/bin/ld: warning: display.o uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.9/../../../arm-none-eabi/bin/ld: warning: lib/libstm32f4.a(stm32f4xx_gpio.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.9/../../../arm-none-eabi/bin/ld: warning: lib/libstm32f4.a(stm32f4xx_rcc.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
arm-none-eabi-objcopy -O binary main.elf main.img
arm-none-eabi-objdump -t main.elf >main.img.sym
cd ../patches/2.032 && make all
make[2]: Entering directory '/home/zarya/md380tools/patches/2.032'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/zarya/md380tools/patches/2.032'
cp ../patches/2.032/patched.img ./experiment.img
# use a fresh experiment.img
cp ../patches/2.032/patched.img ./experiment.img
python2 merge_d02.032.py experiment.img main.img 0x0809D000
Merging an applet.
Loading symbols from main.img.sym
Hooking a menu call.
Merging main.img into experiment.img at 0809d000
../md380-fw --wrap experiment.img experiment.bin
DEBUG: reading "experiment.img"
INFO: base address 0x800c000
INFO: length 0xf2c00
DEBUG: writing "experiment.bin"
make[1]: Leaving directory '/home/zarya/md380tools/applet'

hsmade, 2016-09-12:
Just checked latest firmware and it works for me.
Just to set expectations, these 'rogerbeeps' are the tones that the radio sounds via de speaker when TX-ing. Not via RF!

sijskes, 2016-09-13:
@PD1LOI which or what rogerbeep worked before and how did it sound,  and how does it sound now, or where does it not work?

PD1LOI, 2016-09-13:
Last update was all right. it has apparently been located to my MD-380

sijskes, 2016-09-13:
thanks for reporting, issue closed now.

closed 2016-09-13 by sijskes