travisgoodspeed/md380tools #257: New flashing icon on my MD-390 with latest e800 firmware

iball, 2016-09-05:
Just noticed this.  Icon next to the scan icon now flashes randomly.  This is on an MD-390 (No GPS).
Not seeing it on my MD-380 at all.
Is this normal behavior on an MD-390 now?
See screenshot for details.
[image]

sijskes, 2016-09-05:
flashing 'eye' icons indicates promiscuous mode active.

iball, 2016-09-05:
That's weird because I've never turned it on before on either radio.  
Just found it was turned on by default with the new e800 commit of the code?

sijskes, 2016-09-05:
probably garbage in spi flash, clipped to 1 value = on.

travisgoodspeed, 2016-09-08:
Closing this issue.  Re-open as a new issue if we need to add a checksum to the menu configuration to make it more predictable.

closed 2016-09-08 by travisgoodspeed