travisgoodspeed/md380tools #259: Cygwin Continuous Integration Support

travisgoodspeed, 2016-09-05:
Close this issue when continuous integration builds are being performed for Cygwin, so as not to repeat the trouble that we experienced in #85.

Assigning @m0ivq after he accepted his invitation to join the team.

bendikz, 2016-09-06:
I am using Windows Subsystem for Linux. It worked great for patching the firmware but I haven't figured a way to interface the radio. I assume that it could be possible to mount interfaces since it's a full linux system. I will check on this.

iball, 2016-09-07:
WSfL isn't a full Linux system.  I've tried the same exact thing you did, but it seems the lower-level hardware stuff isn't in there.  
Far easier to just go the "VirtualBox + Linux-distro" route.
That's what I use and I have no problems with it.

m0ivq, 2016-09-07:
Cygwin CI support has now been added to appveyor.yml. Close issue appveyor account is added and functional.

I have moved all of the build commands into ci-script.sh which can be referenced from .travis.yml and appveyor.yml. However, I have left .travis.yml as-is as build seems to hang on clang (gcc is fine).

m0ivq, 2016-09-08:
It appears I just needed to wait a bit longer for clang (OSX) to complete. All changes for CI now merged. Any further changes to the build script used for travis-ci or appveyor should be made in ci-script.sh. Environment changes should still be done in their respective .yml files.

Assigning back to @travisgoodspeed to add an Appveyor account.

travisgoodspeed, 2016-09-08:
I've created an AppVeyor account, and the first build seemed to fail.  The tail end of the log follows.

Re-assigning to @m0ivq to close when the appveyor build is succeeding.

```

24: ID->2502 Country->RU Address->brandmeister.r2aee.net
250997,Parrot Service,Echo Function,,,,Russia,
250999,APRS Gate,APRS Function,,,,Russia,
25: ID->2508 Country->RU Address->hose.r2aee.net
List with special IDs not found!
26: ID->2551 Country->UA Address->master.ur7d.org
List with special IDs not found!
27: ID->2552 Country->UA Address->bm.ham-dmr.com.ua
List with special IDs not found!
28: ID->2601 Country->PL Address->bm.ham-dmr.pl
List with special IDs not found!
29: ID->2621 Country->DE Address->deneb.dstar.su
List with special IDs not found!
30: ID->2622 Country->DE Address->178.238.234.72
List with special IDs not found!
31: ID->2681 Country->PT Address->194.38.140.206
List with special IDs not found!
32: ID->2841 Country->BG Address->193.107.37.135
List with special IDs not found!
33: ID->3021 Country->CA Address->158.69.203.89
List with special IDs not found!
34: ID->3101 Country->US Address->brandmeister.openquad.net
List with special IDs not found!
35: ID->3102 Country->US Address->bm.dmrx.net
List with special IDs not found!
36: ID->3103 Country->US Address->bm.ham-dmr.org
List with special IDs not found!
37: ID->3108 Country->US Address->bmdev.dmrx.net
List with special IDs not found!
38: ID->3109 Country->US Address->bmhose.dmrx.net
List with special IDs not found!
39: ID->3341 Country->MX Address->72.1.241.232
List with special IDs not found!
40: ID->4251 Country->IL Address->46.121.158.50
List with special IDs not found!
41: ID->5021 Country->MY Address->188.166.231.62
List with special IDs not found!
42: ID->5051 Country->AU Address->44.136.151.11
List with special IDs not found!
43: ID->6551 Country->ZA Address->154.66.196.131
List with special IDs not found!
44: ID->7141 Country->PA Address->45.79.177.188
List with special IDs not found!
45: ID->7241 Country->BR Address->104.41.43.207
List with special IDs not found!
curl -L -f 'http://www.dmr-marc.net/cgi-bin/trbo-database/datadump.cgi?table=users&format=csv&header=0' | perl -pe 's,<br/>,,' >>users.csv
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 3088k    0 3088k    0     0  2015k      0 --:--:--  0:00:01 --:--:-- 2297k
curl -L -f 'http://www.dmr-marc.net/cgi-bin/trbo-database/datadump.cgi?table=repeaters&format=csv&header=0' | perl -pe 's,<BR/>,,' >repeaters.csv
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  597k    0  597k    0     0   956k      0 --:--:-- --:--:-- --:--:--  956k
curl -L -f 'http://www.dmr-marc.net/cgi-bin/trbo-database/datadump.cgi?table=users&format=json&header=0' >users.json
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 7077k    0 7077k    0     0  3871k      0 --:--:--  0:00:01 --:--:-- 3871k
curl -L -f 'http://www.dmr-marc.net/cgi-bin/trbo-database/datadump.cgi?table=repeaters&format=json&header=0' >repeaters.json
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 1038k    0 1038k    0     0  1447k      0 --:--:-- --:--:-- --:--:-- 1477k
mv *.zip c:\
rm -r *
cd c:\
mv *.zip %APPVEYOR_BUILD_FOLDER%
cd %APPVEYOR_BUILD_FOLDER%
git fetch
git reset --hard
HEAD is now at 4612c5c moved CI build commands to ci-script.sh
%CYG_BASH% -lc "cd $APPVEYOR_BUILD_FOLDER;./ci-script.sh"
cd firmware && make all
make[1]: Entering directory '/cygdrive/c/projects/md380tools/firmware'
mkdir -p dl
which curl >>/dev/null #Check that we have curl.
curl -f -L $(cat firmware_files.txt | grep "^D002.024.zip" | cut -f2) -o dl/D002.024.zip
make[1]: *** [Makefile:57: dl/D002.024.zip] Error 127
make[1]: Leaving directory '/cygdrive/c/projects/md380tools/firmware'
make: *** [Makefile:20: firmwares] Error 2
Command exited with code 2
```

m0ivq, 2016-09-08:
I have disabled the cygwin tests until I can work out why it has stopped working. Build succeeded prior to pushing, however it doesn't work anymore.

@travisgoodspeed can you please test if build succeeds on appveyor now I have disabled the cygwin tests. I will push a new appveyor.yml once I have it working again.

travisgoodspeed, 2016-09-09:
The most recent build succeeded.  You can view it at https://ci.appveyor.com/project/travisgoodspeed/md380tools

m0ivq, 2016-09-09:
Thanks. Now I know how to RDP into the VM used for the build I'll be able to see more of what's going on without having to keep pushing changes.

m0ivq, 2016-09-13:
It would appear that appveyor rolled back an update to their build worker image on 9th September which may may explain why build was working ok at the time I pushed the commit but not the day after.

travisgoodspeed, 2016-10-17:
Closing this issue, as Appveyor is running reliably.

closed 2016-10-17 by travisgoodspeed